alf45tar/PedalinoMini #1: M5Stack compatability

CraigHissett, 2019-02-13:
https://docs.m5stack.com/#/

The M5Stack (and smaller M5Stick variants) are great ESP32 boards with integrated display.
Would it be be feasible to use this to run the PedalinoMini software?
It would be great to use one as the brains behind a small simple footswitch.

alf45tar, 2019-02-21:
M5Stick could works because has the same OLED but no IO pins. We need to add an I2C port expander.
M5Stack should works without the display.

M5Stack is very interesting but right now is out of the scope of this project. I am focus on different hardware and I will release a new version soon.

Supporting M5Stack is not a top priority of the project but I am open to any discussion.

CraigHissett, 2019-02-21:
Amazing, thanks for the feedback buddy.
I have an i2c expander ordered, and when it arrives I may well have a look at getting this to run on it.
I agree - the M5stack hardware is a great board. One day I'd love to use a few of them in a modular midi build :)

closed 2019-02-21 by CraigHissett