alf45tar/PedalinoMini #2: NodeMCU ESP32 support

MJPees, 2019-05-02:
The NodeMCU ESP32 does not work.
The flash could not be mouted.
Everything else seems to work.
Just the pinning layout is different.
I set the esp32dev environment.

MJPees, 2019-05-07:
Mounting of flash works now.
I had to initialize the flash first.
But if i enable bluetooth I can not open the webconfig.
The ESP reboots...

Serial output:

abort() was called at PC 0x401ec6bb on core 1

Backtrace: 0x400930f0:0x3ffd1300 0x40093321:0x3ffd1320 0x401ec6bb:0x3ffd1340 0x401ec702:0x3ffd1360 0x401da1a7:0x3ffd1380 0x401da3ce:0x3ffd13a0 0x401da385:0x3ffd13c0 0x400eec73:0x3ffd13e0 0x400ef085:0x3ffd1400 0x400ef870:0x3ffd1430 0x400ef885:0x3ffd1450 0x400e5c6a:0x3ffd1470 0x400fd2f1:0x3ffd1490 0x4008ee85:0x3ffd14b0

Rebooting...

alf45tar, 2019-05-11:
Unfortunally 4Mb are not enough to run everything due to limited heap size. WiFi and Bluetooth together are not stable. They can run together for shorter period if you are not using the web interface.
I am trying to fix it using a board with additional 4Mb of PSRAM like TTGO T-Eight ESP32 but due to limited documentation and time available I didn't found the definitive solution.

Cheers

closed 2019-06-15 by alf45tar