alf45tar/PedalinoMini #4: wifi errors and compilation errors

gurnay1973, 2019-06-04:
estimated install vscode and platform.io and then compile and throw many errors. then upload the project and try to connect to the wifi that it generates and tell me without internet connection and I can not enter pedalino.local either. I hope you can help me I find the project excellent. I would like to have it working.

the Projects is charged and passes through the processes mentioned and ends in the creation of the wifi point to which I connect but I can not configure or test it.

Thanks attached photos of the log.

[image]
[image]
[image]
[image]
[image]

gurnay1973, 2019-06-04:
also enter by the .local device name and by the ip too. Seldom if I could connect from sun android, but it opens without styles. Only the html page. The problem is that the pedal device restarts every few minutes and repeats it over and over again. smartconfig, wps and factory reset

Greetings.

gurnay1973, 2019-06-06:
For some reason, the access point created by the pedal is not working. it connects opens the interface but it does not work. This opens without a template and every time it is saved it is restarted. I can not confirm anything, because it restarts every time. I hope you can help me

alf45tar, 2019-06-09:
Did you uploaded file system image?

gurnay1973, 2019-06-10:
Estimated the process you performed was to clone the repository in vscode with platformio. Then change the port and compile, then upload it.
At first I was worried about so many warning in the bookstores. The web panel did not work because it did not load the jquery and popper, bootstrap libraries. Then I blocked the Bootstrap Local variable and started loading the external CDN. So I managed to operate by wps and smartconnect, but the access point even with the problem of loading the local files js and css. Check the code and it fails in the web library config the loading of files through txt.
Now I could use it, but I wish we could fix the local files. Now I also do not know why it reboots from nothing.
Another problem I have is that I have a boss fs6 momentary 2 and when I press it, only the one on the right works, when I press the left the blue light does not shine and a message cc midi does not appear on the screen. I would also like to know how the midi cc are assigned to the momentary 2 and 3 pedals, since I do not understand it yet.
I also need to know what the metronome is for and because when I activate it, the pedal will restart.

Sorry to bother you so much I liked the project and it's the only thing I need to finish it.

Thank you.

alf45tar, 2019-06-10:
- Most of the warning are not linked to Pedalino code but to external libraries Pedalino use. Don't worry about them.
- The local copy of Bootstrap files is stored into SPIFFS. You need to upload the 'data' folder using the following istructions https://docs.platformio.org/en/latest/platforms/espressif32.html#uploading-files-to-file-system-spiffs
- WiFi and Bluetooth together are not stable due to limited heap memory size. They can run together for a short period. Compile WIFI or BLE only. I am trying to fix it using a board with additional 4Mb of PSRAM like TTGO T-Eight ESP32 but due to limited documentation and time available I didn't found the definitive solution yet.
- Boss FS6. Are you using port A, port B or port A&B?
- Momentary 1 use current bank. Momentary 2 use current bank + 1. Momentary 3 use current bank + 2.

gurnay1973, 2019-06-10:
Thank you dear for the information I am clear about local files. Regarding that it restarted, I only uploaded the wifi, since I realized that when compiling the 2, the most frequent reboot occurs. The issue of restart in wifi version is when you are using a lot the website, but it is not so important.
with respect to fs6 I use the port a & by in the setting position momentary 2, but the blue led and the cc note on the screen only appears with the right pedal.
The connection I made for the jack was the one that appears in the diagram. stereo jack to ports a1 - d1 and gnd, incorporating the 3v3 between the a1 and ring. Try changing polarities and I can not make the left work.

Thank you.

gurnay1973, 2019-06-10:
The other query that I almost forgot, is if the latch uses one of the 6 inputs or you can also do something like the FS6.

Thank you.

MJPees, 2019-06-11:
Wifi without ble is also not stable. If you navigate between the pages of the  webconfig the webserver crashes. I think it is a problem with async webserver itself.

gurnay1973, 2019-06-11:
I also believe the same. Friend you could place a fs-6 as momentary 2, because I have not managed to operate both pedals and connect them to a & b.
The connection I made for the jack was the one that appears in the diagram. stereo jack to ports a1 - d1 and gnd, incorporating the 3v3 between the a1 and ring. Try changing polarities and I can not make the left work.

alf45tar, 2019-06-11:
@MJPees webconfig is very stable for me apart the Remote Display.
Remote Display is not fundamental and I consider it a bonus feature. In general all the Live tab is not finished yet. I used it for learning websocket.

gurnay1973, 2019-06-11:
???? help me please

alf45tar, 2019-06-11:
@gurnay1973 I don't own any FS-6 but you can help me to find the fix.
Can you confirm that when you press the left pedal the tip is shorted to GND and when you press the right pedal the ring is shorted to GND (or viceversa)?

gurnay1973, 2019-06-11:
do you mean the jack that solves the arduino or the jack that is inside the factory pedal?

alf45tar, 2019-06-11:
the jack inserted into A&B port

gurnay1973, 2019-06-11:
because I connect a momentary only without the a2 and it works as momentary 1. The only thing I did afterwards was to connect the a2 to the resistance and then to the 3v3, then to the ring. I connect the FS6 and the right one works.

gurnay1973, 2019-06-11:
I'll try doing the manual crossover

alf45tar, 2019-06-11:
Forget Pedalino for a moment.

- Connect a jack into A&B port
- Press left pedal and check if tip (or ring) is shorted with sleeve
- Press right pedal and check if ring (or tip) is shorted with sleeve

Let me know your finding

gurnay1973, 2019-06-11:
right tip
left ring

gurnay1973, 2019-06-11:
if I connect them separately to the digital port. they work both separately, if I connect them to analog and digital together, only the right one works.

gurnay1973, 2019-06-14:
?????

alf45tar, 2019-06-15:
Are you sure you connected the correct analog pin?
What Pedalino call A1 is marked VP on DOIT board and it is connected to GPIO36.

gurnay1973, 2019-06-15:
connect to the sp that on my plate corresponds to 36, and also probe with the configuration 3 that is with g34 and g27. it does not work either

gurnay1973, 2019-06-15:
If you can, there will not be a more fluent way to chat like WhatsApp or Hangouts. Thank you

gurnay1973, 2019-06-24:
??????

alf45tar, 2019-08-19:
Did you solved? Any improvement?

closed 2019-08-23 by alf45tar