alf45tar/PedalinoMini #5: Switch from WiFI Mode to BLE mode

Hans68, 2019-06-12:
The last update of platformio.ini includes  two different environments for WIFI and BLE and it is loaded successfully. 
But how can I switch in operation from WiFi to BLE and vice versa? 
Or if I have to load two different versions from the beginning due to limited 4MB, how do I have to modify latest versio of platformio.ini?
Thanks in advance for your help.

MJPees, 2019-06-13:
Hi, I think it is currently not possible to switch the mode at runtime.
You have to compile with different build_flags or enable/disable  -D NOBLE and -D NOWIFI

Hans68, 2019-06-13:
Thanks a lot for your quick response. What you suggested worked fine with the previous platformio.ini, but I tried this with the newest version, it finally  boots always in WiFi mode.
There are 
Build-flags esp32
Build-flags esp32 Ble
Build-flags esp32 WiFi
Where to comment/uncomment NOBLE respectively NOWIFI.
Thanks a lot in advance! 

Greetings 
Hans

MJPees, 2019-06-13:
Ok, I think you use the env:esp32...

So you should replace the 
build_flags = ${common.build_flags_esp32} 
by
build_flags = ${common.build_flags_esp32-ble}

[env:esp32dev]
platform 	= espressif32
framework 	= arduino
board 		= esp32dev
build_flags = ${common.build_flags_esp32-ble}
....

Alternative you can set -D NOBLE as comment by adding the ";" at start of line next to -D NOBLE at the build_flags itself. (The text should get green at VSCode)
To deactivate wifi remove the ";" in front of -D NOWIFI.

Greetings
Marc

Hans68, 2019-06-14:
Thank you very much!I will be out for some days and will give you a feedback as soon as I can test it.Greetings Hans Von Samsung-Tablet gesendet

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/alf45tar/PedalinoMini/issues/5?email_source=notifications\u0026email_token=ALR3NU6XC26NL3ONWRJGMVTP2LAV5A5CNFSM4HXKRGIKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODXVFPZY#issuecomment-501897191",
"url": "https://github.com/alf45tar/PedalinoMini/issues/5?email_source=notifications\u0026email_token=ALR3NU6XC26NL3ONWRJGMVTP2LAV5A5CNFSM4HXKRGIKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODXVFPZY#issuecomment-501897191",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

alf45tar, 2019-06-15:
No changes are required to platformio.ini in order to upload Wifi or BLE version.
Just go to PlatformioIO Project Task and select Upload under env:esp32doii-devkit-v1-wifi or env:esp32doii-devkit-v1-ble.

closed 2019-06-21 by alf45tar