alf45tar/PedalinoMini #7: Build and Upload

Hans68, 2019-06-27:
First of all congratulations to this great project !
I have following questions:
1. Do I have to build and upload always first the complete code including file system image or is it enough to build and upload only either wifi or BLE ? If I upload only wifi I get an error message when uploading File Systeme Image of wifi, . Nevertheless it works, if I load the file system image of the complete package.
2. Is it be possible to switch between BLE and wifi in operation in case of using ttgo-t-eight?
This would be worth the investment.

alf45tar, 2019-06-27:
1. It is enough to build and upload wifi or ble. The complete build and upload is if you want to try both together. I will check the error for Upload System Image under wifi.

2. No, it is not possible. ttgo-t-eight is an all-in-one board with a bigger display, the battery interface and 3 buttons. PSRAM is not yet used.

Hans68, 2019-06-27:
Thanks a lot, understood, but how can I try wifi and BLE togehter, I learnt only one option will work at the same time?

alf45tar, 2019-06-27:
It works but it is not stable due to heap memory size. You can upload wifi firmware, make the configuration, upload wifi and ble, use it. Do not open the web config interface with wifi+ble otherwise it will reboot.

Hans68, 2019-06-27:
Great response time !
My preferred solution would be:
Only BLE Midi (NoWiFi Midi, no USB midi etc.)
Configuration via WiFi or USB Interface (Node MCU -USB- Interface) with new flashing.
This would help to configurate and afterwards operate it in BLE without new flashing.
Is there anything planned for the future?

alf45tar, 2019-06-27:
The next improvements will be:
- Replace Arduino Uno with Arduino Pro Mini for USB MIDI
- Switch between WIFI and BLE without firmware upload. I don't know if makes sense because you have to link both libraries (WIFI and BLE) and I don't know if there is a real heap memory saving.

Hans68, 2019-06-27:
Thank you!
Sorry, I meant of course
Configuration via WiFi or USB Interface (Node MCU -USB- Interface) with**out** new flashing.

alf45tar, 2019-07-14:
The latest improvments can solve your issue.

Hans68, 2019-07-14:
Thank you !!
I have been testing the different Build options. Running the overall Build I got several error messages, see attached copy.
If I run the specific Builds, everything is goining fine without error.
I had no time yet to test upload. What is esp32dev for resp. ttgo-t-eight ? Is this the combined function of WiFi and BLE ? Is it functioning now?
What's about "Upload File System Image" ? Is it substituted by "Upload and monitor" ?
[image]

alf45tar, 2019-07-14:
Did you updated to Plafformio Core 4.0?
esp32dev is for generic esp32 board with wifi+ble
esp32doit-devkit-v1 is for doit-devkit-v1 board with wifi+ble
esp32doit-devkit-v1-ble is only ble
esp32doit-devkit-v1-wifi is only wifi
ttgo-t-eight is for ttgo-t-eight board with wifi+ble
ttgo-t-eight-ble is ony ble
ttgo-t-eight-wifi is only wifi

If you don't have doit or ttgo board I suggest esp32dev.
wifi+ble is working fine is you are not using webconfig.
I suggest to boot in wifi mode (istructions are in the README), make configuration via webconfig and reboot in wifi+ble

Cheers

Hans68, 2019-07-14:
Thanks, your response time is outstanding !!
Yes, I have Platformio Core 4.0.0 and I have esp32doit-devkit-v1. 
I wait to receive ttgo-t-eight within the next days. 
Cheers

Hans68, 2019-07-25:
Hi,
I am back for testing !
With the latest build I had problems with the configuration files in the browser window, see attached copy. The same effect I had with older versions when I forgot to "upload file system image". 
The a.m. behaviour happens with both HW versions, with ESP32 dev. kit and  with TTGO-T-EIGHT.
If I go back to the version, which you can see in the second attached picture, configuration file in Browser is as it should be.
In addition I have a new effect, which is most probably independant of your build versions, but it surprises me: On my ESP32 Dev. Kit I have to press Boot button when "Connecting ....." appears on the PIO terminal window. This was not necessary before. Perhaps you have a hint for me.
Best regards
Hans
[image]
[image]

alf45tar, 2019-07-29:
Did you try to reset it to factory default?
I recently moved from EEPROM library to Preferences library for saving configuration.

Hans68, 2019-07-30:
I tried everything you recommended and I used ESP32-Dev. Kit and TTGO-T-Eight, still the same problem: Config Files are shown in browser as if 
[image]
I wouldn't have uploaded File System Image.
(see attached photo)
Cheers

alf45tar, 2019-07-30:
It looks like that bootstrap files are not present. Did you uploaded File System Image?
Click Options and try a different WebUI Theme.

Hans68, 2019-07-30:
Yes as already mentioned, file system image is loaded and if I go back to a previous build (see my comment 5 days ago), config tables are shown correctly.
That is why I don't believe that it is a browser issue.

alf45tar, 2019-07-31:
Did you try to change the WebUI Theme under Options?

alf45tar, 2019-07-31:
Can you send a screenshot of Chrome Inspector? I think there are some errors there.

Hans68, 2019-07-31:
I got some problems as I am using German version of Chrome, so I can not
directly find these wordings like WebUI and Chrome Inspector,
but: It worked before with my I-phone browser, my Android browser, and with
Fire Fox. Now, non of these Browsers are working!

Hans68, 2019-07-31:
I just recognized, that with the new builds, I have no "Device:" shown on the display. With the old version (end of june) which I tried for testing again, I get a "Device: 2512CF4A". As already told, with the version where I get the Device Name, Config Files in Browser are o.k..
Please see attached photos. 

[image]
[image]
[image]

alf45tar, 2019-08-01:
Can you send me the output in the serial console during boot?

Hans68, 2019-08-01:
As I am not yet very skilled with Platformio, please give me some more
detailled steps what to do.
I will be out of home now for 4 days and can be back on Monday afternoon.
Thanks and Cheers

Hans68, 2019-08-06:
Are these the printouts you asked for ??
"SW Version working" is app from end of June
"SW version not working" is the newest build
[SW Version 5 working.pdf](https://github.com/alf45tar/PedalinoMini/files/3471387/SW.Version.5.working.pdf)
[SW Version new not working.pdf](https://github.com/alf45tar/PedalinoMini/files/3471388/SW.Version.new.not.working.pdf)

alf45tar, 2019-08-06:
Thank you for the console messages log.
I think your problem is related to Non Volatile Storage initialization.
I recently moved from EEPROM library to Preferences library for saving configuration but it request a reset it to factory default.

Here the procedure: https://github.com/alf45tar/PedalinoMini#booting-modes

I will fix my code to perform a automatic NVS initialization on first boot.

Thanks
alf45tar

Hans68, 2019-08-06:
Thanks !I did already Reset to Factory default last week with no result. I tried it before and after upload. At what status do you recommend to do it?But I think best is to wait for your update.CheersHansVon Samsung-Tablet gesendet

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/alf45tar/PedalinoMini/issues/7?email_source=notifications\u0026email_token=ALR3NU5DESBS3CGUW2GIMH3QDFNTVA5CNFSM4H3Z7TKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3U2LHQ#issuecomment-518628766",
"url": "https://github.com/alf45tar/PedalinoMini/issues/7?email_source=notifications\u0026email_token=ALR3NU5DESBS3CGUW2GIMH3QDFNTVA5CNFSM4H3Z7TKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3U2LHQ#issuecomment-518628766",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

alf45tar, 2019-08-07:
I just uploaded the requested updates. Try again and let us know.

Hans68, 2019-08-08:
Unfortunately no improvement ! Please find attached printout.

Cheers

Hans68, 2019-08-08:
Also no improvment with ttgo-t-eight concerning this issue.

alf45tar, 2019-08-08:
> […](#)

There is no attachment

alf45tar, 2019-08-19:
Did you solved?
If not check the log during reset to factory default. It is clear indicated if the reset has been completed.

Hans68, 2019-08-19:
No, it is not solved as already stated on August, 7th.
Please find attached log during reset to factory default.
Cheers

Hans68, 2019-08-19:
Just checked latest build: Latest commit 324cb4e 21 hours ago
This is not working, neither with ESP32 dev kit nor with ttgo-t-eight. 
I only get a display showing an E, sometimes quickly changing to F, see attached photo.
[image]

alf45tar, 2019-08-19:
Sorry but there is no attachment.

The E and the F are ok. The 6 pedals are named A to F by default.
If E or F is appering suddently is because there is no pull-resistor as shown is Schematic Sheet 2.

Hans68, 2019-08-20:
Thanks, 
I only had pull-up resistors on VP and VN as shown in "PedalinoMini_bb.png". This picture shows also my test configuration.
The additional pull-up resistors solved the problem of displaying E/F, but the problem of not showing the device name and the configuration files in the right way remains.
Attached the log after reset. 
[2019-08-20 dev kit wifi.pdf](https://github.com/alf45tar/PedalinoMini/files/3519401/2019-08-20.dev.kit.wifi.pdf)

alf45tar, 2019-08-20:
PedalinoMini_bb.png is an old image. I will delete it soon.

I don't need the log after reset but the log during the reset. It should looks like this one:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2-18-g977854975
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 233496, Internal Free Heap 206724
PSRAM Total Heap 4194252, PSRAM Free Heap 4194252
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/
                                                                                  (c) 2018-2019 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 8E2DE6B4
Bluetooth Classic disabled: 15348 bytes released
Internal Total Heap 248880, Internal Free Heap 222072
PSRAM Total Heap 4194252, PSRAM Free Heap 4194252
######################################################################################################################
Reset EEPROM to factory default
Updating NVS ...
done
[NVS][Global][Device Name]: 8E2DE6B4
Updating NVS ... done
[NVS][Global][SSID]:
[NVS][Global][Password]:
Updating NVS ... done
[NVS][Global][Bootstrap Theme]: bootstrap
Updating NVS ... done
[NVS][Global][Current Profile]: 0
Updating NVS ... done
[NVS][Global[Bank Switch]: 0
Updating NVS ... done
[NVS][Global[Blynk Cloud]: 0
Updating NVS ... done
[NVS][Global][Blynk Token]:
Updating NVS Profile A ... done
Updating NVS Profile B ... done
Updating NVS Profile C ... done
Reading NVS Global ... done
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START
SmartConfig started

Hans68, 2019-08-20:
I hope this is what you need

[2019-08-20 dev kit wifi reset.pdf](https://github.com/alf45tar/PedalinoMini/files/3522297/2019-08-20.dev.kit.wifi.reset.pdf)

alf45tar, 2019-08-20:
You never reset to factory default your board. I don’t see the following lines in your log.

######################################################################################################################
Reset EEPROM to factory default

Please follow the procedure or let me know what is not clear.

Hans68, 2019-08-20:
I did this Reset to Factory default several times, but I am not sure if it is important to do it during the reset process is running. Give me a hint please when to do it.

alf45tar, 2019-08-21:
Reset to factory default 

DO IT board
Press and **release** EN button and immediately after press and **hold** BOOT button.

TTGO board
Press and **release** POWER button and immediately after press and **hold** CENTER button.

In both cases keep pressed BOOT button at least 15 seconds or look into log for wording "Reset EEPROM to factory default"

Hans68, 2019-08-21:
Thanks, solved !!
I always released the boot button when "reset to factory default" appeared in the OLED display. Your hint "Keep pressed until the progress bar reach the end. If you release the button before the progress bar reach the end Pedalino will start in on of the supported boot mode" 
I recommend to add this sentence under "Booting Modes".

[2019-08-21 reset to factory default.pdf](https://github.com/alf45tar/PedalinoMini/files/3524015/2019-08-21.reset.to.factory.default.pdf)

closed 2019-08-21 by Hans68