alf45tar/PedalinoMini #8: uploadfs

mknerr, 2019-07-15:
Thanks so much for all your work on this project! Very excited to get this working, and it seems I'm close. I'm just not following on what I need to do to upload file system. Can't find command and everything I've tried in the terminal gets an error that it is not a recognized command. I'm sure this is simple, but all of the help I've searched for is just not doing it for me. Thanks again!

alf45tar, 2019-07-16:
Edit default_envs variable in platformio.io and leave one env only.
If you have a DOIT board it should be:
default_envs = esp32doit-devkit-v1

default_envs

Run “Upload File System image” task in PlatformIO IDE

mknerr, 2019-07-16:
Thanks for the reply. My board is actually the Heltec WiFi Kit 32. Is that going to complicate things? I did attempt to put that in the .ini, but I'm not sure I got it right. I had the Heltec and the "generic" both as default though, so that could have been part of it. 

I did fix the com port. 
Thanks again!

mknerr, 2019-07-16:
This is what I've added to the end of the .ini:
`[env:heltec_wifi_kit_32]
platform     = ${env.platform}
board        = heltec_wifi_kit_32
build_flags  = ${common.build_flags_heltec_wifi_kit_32}
upload_port  = COM3
monitor_port = ${env:heltec_wifi_kit_32.upload_port}`

Not sure if I'm missing something, such as flags, though.

alf45tar, 2019-07-16:
You can use the following platformio.ini file

`
[platformio]
default_envs = heltec-wifi-kit-32

[common]
build_flags_esp32 =
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=2
	-D WEBSOCKET
	-D DIAGNOSTIC
;	-D NOWEBCONFIG
;	-D NOBLYNK
build_flags_esp32-ble =
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=2
	-D NOWIFI
build_flags_esp32-wifi =
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=2
	-D NOBLE
	-D WEBSOCKET
	-D DIAGNOSTIC
;	-D NOWEBCONFIG
;	-D NOBLYNK

[env]
platform 	           = espressif32
framework 	           = arduino
board 		           = esp32dev
board_build.f_flash    = 80000000L
board_build.partitions = min_spiffs.csv
lib_deps			   = AppleMIDI
						 Battery Sense
						 Blynk
						 Bounce2
						 AsyncTCP
						 ESP Async WebServer
						 ESP8266 and ESP32 Oled Driver for SSD1306 display
						 MD_UISwitch
					  	 https://github.com/alf45tar/arduino_midi_library
					  	 OSC
					  	 RemoteDebug
						 ResponsiveAnalogRead
						 https://github.com/lathoub/Arduino-BLE-MIDI
monitor_speed	       = 115200

[env:heltec-wifi-kit_32]
platform 	 = ${env.platform}
board 		 = heltec_wifi_kit_32
build_flags  = ${common.build_flags_esp32}
upload_port  = COM3
monitor_port = ${env:esp32dev.upload_port}

[env:heltec_wifi-kit-32-ble]
platform     = ${env.platform}
board 		 = heltec_wifi_kit_32
build_flags  = ${common.build_flags_esp32-ble}
upload_port  = COM3
monitor_port = ${env:esp32doit-devkit-v1-ble.upload_port}

[env:heltec-wifi-kit-32-wifi]
platform     = ${env.platform}
board 		 = heltec_wifi_kit_32
build_flags  = ${common.build_flags_esp32-wifi}
upload_port  = COM3
monitor_port = ${env:esp32doit-devkit-v1-wifi.upload_port}
`

mknerr, 2019-07-16:
Thanks so much!  I'll give it a shot.

mknerr, 2019-07-16:
Seems like I'm up and running! Only issue that I know of at this point is the OLED is not functioning. Not sure if that would be a driver issue or what. 
Very happy to get to this point, though. Now I can look into what I want to do on the hardware side of things.

alf45tar, 2019-07-16:
OLED is not working because according to https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_kit_32/pins_arduino.h 

static const uint8_t SCL_OLED = 15;
static const uint8_t SDA_OLED = 4;

and not SDA 21 e SCL 22.

Not hard to fix. Change from  

#define OLED_I2C_SDA      SDA
#define OLED_I2C_SCL      SCL

to 

#define OLED_I2C_SDA      SDA_OLED
#define OLED_I2C_SCL      SCL_OLED

in https://github.com/alf45tar/PedalinoMini/blob/master/src/DisplayOLED.h

May be is not the only changes.

mknerr, 2019-07-17:
Thank you. I made the changes, but didn't seem to do the trick.

alf45tar, 2019-07-17:
The OLED connection is not "standard". There is a customized library for it here https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/src/oled

The main difference is the OLED_RST pin. Try to add the following code before display.init  
https://github.com/alf45tar/PedalinoMini/blob/57af0861bda4cdfa207430fc4564c3e14b532290/src/DisplayOLED.h#L870

   		pinMode(16,OUTPUT);
    		digitalWrite(16, LOW);
    		delay(50);
    		digitalWrite(16, HIGH);

mknerr, 2019-07-17:
Thank you! That did wake it up. Although, it seems like it needs to be shifted down about 10-15 pixels. 
Would you recommend I replace the OLED filles from the heltec github, or is it not that simple? I'm guessing it's not. 

 Also getting some random activity of pedals 5 and 6 when I touch the board, but i figured its some sort of grounding issue since nothing is attached yet.

Thanks again for all of your help.

mknerr, 2019-07-17:
So, just a quick update, I rebooted, actually a full reset, and when it came up the vertical shift in the OLED had changed. It's about 50% off now.

alf45tar, 2019-07-18:
Replacing OLED files from the heltec github should works.

Pedal 5 and 6 are defined as analog in the default configuration. In this case a pull-up resistor is mandatory on GPIO 33 and GPIO 32 to avoid spurious reading. Just for testing you can connect pins to GND.

Are you using the default pin assignments or you remapped pins?

`const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_14, GPIO_NUM_12, GPIO_NUM_13};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_39, GPIO_NUM_34, GPIO_NUM_35, GPIO_NUM_32, GPIO_NUM_33};`

alf45tar, 2019-07-18:
...or use a connector with a switch.  Some connectors have a switch that can ground your analog input when there is nothing plugged in, but will disconnect the analog input from ground when something is plugged in.  

This may cost a little more than a resistor but does not load the power.

closed 2019-08-03 by alf45tar

jaylfc, 2020-03-01:
Hi, sorry to comment on an old thread but I tried the above platformio.ini file and get immediate parser errors. Could you please help?

mknerr, 2020-03-02:
That’s a pretty old ini. You might try just adding the last three sections to the current ini, and also replace your first line with the first line in the ini above (default_envs = heltec-wifi-kit-32). 

Also, I am not using this board anymore so I won’t have much more insight.

jaylfc, 2020-05-31:
> Replacing OLED files from the heltec github should works.
>
> Pedal 5 and 6 are defined as analog in the default configuration. In this case a pull-up resistor is mandatory on GPIO 33 and GPIO 32 to [...]

How would I go about replacing the OLED files, I have everything working but the screen on a Heltec board and it’s driving me insane.

Could anyone offer some insight, thanks!