alf45tar/PedalinoMini #9: No Battery Indication

Hans68, 2019-07-28:
I have connected a 18650 Lion Battery to TTGO-T-EIGHT and the operation and charging is fine, but I have no Battery indication, neither in text (Battery voltage: 0mV)  nor in symbol.
Is it not implemented yet ??
Please also have a look to my other issue, I have no usable configuration picture in browser any more with the current build, it is as if I wouldn't have uploaded file system Image.
Thanks a lot!
Cheers

alf45tar, 2019-07-29:
Battery indication on the top left of OLED is available only on TTGO-T-EIGHT build.

The battery voltage is for testing only because there are no more available GPIO.
It works connecting GPIO34 in the center of 2 resistors divider (10K each) because you cannot measure directly 3.7V (or more up to 4.2V).

https://github.com/alf45tar/PedalinoMini/blob/57af0861bda4cdfa207430fc4564c3e14b532290/src/Pedalino.h#L36

Hans68, 2019-07-30:
Thanks a lot for your hints!!
I connected GPIO 34 as you recommended and battery indication in the right upper corner is working fine, but text is showing 14.226 mV instead of app. 3.7 mV. Looks like a factor of 4 is calculated. (see attached photo)
[image]

alf45tar, 2019-07-30:
You cannot measure 3.7V using a GPIO pin. GPIO pin are limited to 3.3V.
For this reason I am using a [divisor ](https://github.com/rlogiacco/BatterySense#voltage-divider-ratio) in order to measure the half of the value. The software multiply the value read by 2.

Please also take note that the value of a Lion battery can reach 4.2V during charging.

Hans68, 2019-07-30:
This was clearly understood and there are only app. 1.9V at GPIO 34. I only wanted to inform you, that the battery voltage shown in figures is 14226mV instead of 3700 mV . (See my picture)

Hans68, 2019-08-08:
With latest build for ttgo-t-eight solved !
Thanks and closed !

closed 2019-08-08 by Hans68