alf45tar/PedalinoMini #10: Pedal 5 reading input change with nothing connected

jberry, 2019-08-01:
My newly flashed ESP32's screen is a weird white mess, so I connected a serial monitor and saw something odd: it seems that it keeps seeing some input change for Pedal 5, even though nothing is connected to that pin.  Not sure if that's the cause of my display issue, but it's definitely an oddity.

Any ideas?

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2-18-g977854975
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 330232, Internal Free Heap 303500
PSRAM Total Heap 0, PSRAM Free Heap 0
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2019 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 2412CFA4
Reading NVS Global ... done
USB MIDI started
DIN MIDI started
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
SmartConfig started
SmartConfig timeout
WPS started
WPS timeout
SYSTEM_EVENT_STA_STOP
AP Pedalino-2412CFA4 started with password 2412CFA4
AP SSID     : Pedalino-2412CFA4
AP PSK      : 2412CFA4
AP MAC      : A4:CF:12:24:52:31
AP IP       : 192.168.4.1
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
Channel     : 1
Connect to Pedalino-2412CFA4 wireless network with password 2412CFA4
mDNS responder started
OTA update started
E (33529) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:52] begin(): Mounting SPIFFS failed! Error: -1
HTTP server started
Connect to http://2412CFA4.local/update for firmware update
Connect to http://2412CFA4.local for configuration
ipMIDI server started
RTP-MIDI started
OSC server started
Internal Total Heap 324568, Internal Free Heap 202696
Loading profile ...
Reading NVS Profile A ... done
Bank  1
Pedal  1     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF     60   Channel  1   Pin D25
Pedal  2     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF     62   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF     64   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF     65   Channel  1   Pin D14
Pedal  5     MIDI         ANALOG                     POLARITY-   CONTROL_CHANGE  12   Channel  1   Pin A32 D12
Pedal  6     MIDI         ANALOG                     POLARITY-   CONTROL_CHANGE  13   Channel  1   Pin A33 D13
Pedal  5 calibration min 36
Pedal  5   input 36 output 0 velocity 0.00
Pedal  6 calibration min 0
Pedal  6 calibration max 0
Pedal  6   input 0 output 0 velocity 0.00
Pedal  5   input 40 output 127 velocity 15.20
Pedal  6   input 0 output 127 velocity 31.62
Pedal  5 calibration min 20
Pedal  5 calibration min 0
Pedal  5 calibration max 0
MTC None
Pedal  5   input 58 output 0 velocity -0.59
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
Pedal  5   input 0 output 127 velocity 231.33
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Pedal  5   input 44 output 0 velocity -7.77
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
Pedal  5   input 0 output 127 velocity 236.06
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Pedal  5   input 8 output 0 velocity -7.15
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
CONTROL CHANGE.....Code  12......Value   0.....Channel  1
Pedal  5   input 0 output 127 velocity 233.46
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Pedal  5   input 9 output 0 velocity -7.05`

jberry, 2019-08-02:
I fixed my display with 2 things:
I'm running an 1106 on an ESP32, so I had to modify DisplayLED to load the 1106 drivers.
The schematic on the README has the SDA line going to a pin labeled SCL, and vice-versa, so I got confused what went where.

Not sure if I have a bad devkit or what, I switched to a second one I had and it's working as-expected.  

I'll leave this open so you see the schematic error, but feel free to close this after you see that/acknowledge

alf45tar, 2019-08-02:
You cannot leave GPIO floating (not connected). You need a pull up resistor as indicated in schematic sheet 2.
By default pedal 5 and 6 (GPIO 32 and 33) are setup as analog input. The pull up resistor is mandatory when not connected to anything.
For digital input (momentary pedal) is not usually mandatory.

alf45tar, 2019-08-03:
Schematic fixed
Thanks

closed 2019-08-03 by alf45tar