alf45tar/PedalinoMini #12: 3V3 Logic on ESP device

prisme60, 2019-08-15:
Hello Alf45tar,

First Thank you for your project. I was thinking about a similar project than yours. My goal is only MIDI IN / MIDI OUT from / to Wifi network midi.

I am a professional software programmer, but I'm **not** an electronic engineer. I see 6N137 (High Speed Optocoupler) is powered by 5 V (the Vcc seems OK for me), but the pin 6 (Vo) of the 6N137 component is directly connected to the ESP. 

I suppose that you "transpose" the electronic of the Pedalino project to the Pedalino mini, and you forget that Arduino HWs are 5V, but ESPs are 3V3 and **not 5V tolerant**. 

I think you need to adapt the output of the pin 6 (Vo) of the 6N137. If you have an oscilloscope, it should be possible to check it.

alf45tar, 2019-08-16:
Technically speaking you are right. Practically speaking it works.

I used the 6N137 coming from Pedalino. As you noted it should be replaced by somenthing that works at 3.3V like a 6N138. If you want to propose and improved circuit it will be appreciated.

prisme60, 2019-08-19:
Which schematic editor are you using? In your github project, you just provide the SVG exports. I will be easier to me to start with your schematic project.

Before submit an improved circuit, I need to test it myself (I am still waiting to receive DIN connectors), and I record you that I am a "newbie" in electronics ("my" answer will be built from solution found by others on the Internet).

alf45tar, 2019-08-19:
I am using EasyEDA. Here attached the link the project source file.

https://easyeda.com/editor#id=|29d4a1b85ae64276ad0eff88e8fdee0d|e760a5d776174bcda2c870520d9c9332|dfcbfd3e075647c3b81704f302e61d13|8744b22c13f94633bc80251eca18fb8f

prisme60, 2019-08-23:
**Here is a draft proposal. Be very critical on the schematic.  It is not tested for the moment. I reject all responsabilities of possible hardware destruction. Be aware, it is not my formation.**

[https://easyeda.com/christian.foucher/pedalinomini-3v3-with-6n138-and-bss138](https://easyeda.com/christian.foucher/pedalinomini-3v3-with-6n138-and-bss138)

- According to the midi specification, the MIDI OUT connector side shall be connecteted to the GND : [Midi DIN electrical specification](https://www.midi.org/specifications-old/item/midi-din-electrical-specification)
- I have replaced 6N137 by 6N138 like it was done on the following project : [Midi module assembly instructions](http://10rem.net/blog/2012/03/31/midi-module-assembly-instructions)
- I have added a resistor on the pin 7 of the 6N138, because it improves the response time by better rising and falling edges : [MIDI for the Arduino Circuit Analysis YouTube](https://www.youtube.com/watch?v=0L7WAMFWSgY) at the time 00:13:44 of the video (but you should see all the video, it is very well explained).
- I have added a level shifter with the BSS138 component, like we can find on several electronic level shifters : [Sparkfun](https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all), [Adafruit](https://www.adafruit.com/product/757)
- I have changed the perimeter of the optional zone for the led of the MIDI IN, because the resistor is not optional.

[Schematic_PedalinoMini-3V3-with-6N138-and-BSS138_Sheet-0_20190823090030.svg.zip](https://github.com/alf45tar/PedalinoMini/files/3533368/Schematic_PedalinoMini-3V3-with-6N138-and-BSS138_Sheet-0_20190823090030.svg.zip)

Remarks : You can use the "MC74VHC1GT08" (2-Input AND Gate/CMOS Logic Level Shifter) like it was proposed in [http://10rem.net/blog/2012/03/31/midi-module-assembly-instructions](midi-module-assembly-instructions) instead of the BSS138 level shifter if you prefer but it seems to be more difficult to buy.

marosell, 2019-08-30:
Apologies, I am neither a professional programmer or electronics designer. However, I see that power coming into the MIDI-IN optional zone is now 3.3V but the power coming into the MIDI-OUT optional zone remains 5V. Were you not proposing that the entire system move to 3.3 as the ESPs are not necessarily designed to be 5v tolerant? The screen runs on 3.3 anyways, not sure why it's worth keeping track of a second voltage?

There is also an old reference to 5V on the ESP32 pin 15.

alf45tar, 2019-08-30:
> The screen runs on 3.3 anyways, not sure why it's worth keeping track of a second voltage?

Good point. I think the MIDI-OUT port can be converted to 3.3V replacing the 220 ohm resistors with a 10 and a 33 ohm resistors. This solution can avoid the level converter.

I will update my schematic with 6N138 and 10/33 resistors.

marosell, 2019-08-30:
Ugh. Actually, I'm not sure you can do this, as Midi is a 5v protocol, so on output other devices are expecting 5v. Does that sound right? I don't even know.

prisme60, 2019-08-30:
> There is also an old reference to 5V on the ESP32 pin 15.

I'm agree, The ESP32 should not be powered by 5V.

> I think the MIDI-OUT port can be converted to 3.3V replacing the 220 ohm resistors with a 10 and a 33 ohm resistors. This solution can avoid the level converter.

Like Marosell, I think adaptation of voltage level is needed. For the moment, I can't test it for the moment, I have all components, except my 5-pins DIN connectors and USB-MIDI converter (I have been waiting for 75 days! it seems to be lost during shipping!). I don't want to start the test with my Yamaha CLP120 keyboard.

alf45tar, 2019-08-30:
According to the following schematic https://www.dropbox.com/s/jefwxxtufgwg0ex/esp32_Schematic%20Prints.pdf?dl=0 you can apply 5V to Vin. It is an alternative power source instead of USB.

However my intention is not to power supply ESP32 using Vin pin but it is to drain 5V from Vin for MIDI. PedalinoMini need to be connected to an USB power source (or battery) to power all the accessories.

MIDI has been recently update to works with 3.3V and now 3.3V is on the standard. The value of the 2 resistors (10 and 33 ohm) for a MIDI-OUT are in the standard schematic (https://forum.arduino.cc/index.php?action=dlattach;topic=566442.0;attach=272586).

prisme60, 2019-08-30:
>According to the following schematic https://www.dropbox.com/s/jefwxxtufgwg0ex/esp32_Schematic%20Prints.pdf?dl=0 you can apply 5V to Vin. It is an alternative power source instead of USB.

You can apply 5V to Vin, but the Vin is not directly connected to the ESP32, there is the voltage regulator (NCP1117) that transforms 5V to 3.3V and gives the signal labeled VDD3V3 on pin2 of the ESP32.

prisme60, 2019-08-30:
> MIDI has been recently update to works with 3.3V and now 3.3V is on the standard. The value of the 2 resistors (10 and 33 ohm) for a MIDI-OUT are in the standard schematic (https://forum.arduino.cc/index.php?action=dlattach;topic=566442.0;attach=272586).

Thank you for your information source. I will try when I have all the hardware.

Psychlist1972, 2019-08-30:
Hi All. Mike contacted me and pointed me here.

MIDI over DIN is a 5v standard. There are times when you can get 3.3v to work, but that is highly dependent upon the receiving end. It will **not** work with all devices. You're relying on the receiving end's opto isolator to trigger at 3.3v logic levels, which is not guaranteed. There are also lots of bad MIDI references out there. There are also synthesizers and other small devices that are inline MIDI, with no other power supply, and which rely on the 5v. It's questionable as to whether or not those are in-spec, but they exist.

The IC I used for level shifting was picked specifically because of the voltage ranges it supports for in/out. It's running at 5v, but the input signal is based on that from a 3v3 MCU. So it needed to be able to take a low enough ON value (3v3 logic level). There may be others than can be substituted, but I didn't find any at the time. You'll need to compare the specs for minimum "on" signal. It's a small SMD package, but it's easy to hand-solder if you leave large enough pads in your design.

Here's the one I used. If you look at page 3, you can see it will trigger at 2v when supplied with between 4.5 and 5.5 volts VCC.
https://www.onsemi.com/pub/Collateral/MC74VHC1GT08-D.PDF

If you want to use this for MIDI out, all you need is
- +5v to MIDI through a 220R on Pin 4 on the MIDI DIN
- The signal through the level shifter and a 220R on Pin 5 of DIN
- Ground on Pin 2 of DIN

Basically, just the bottom half of my schematic. This:

[image]

Good luck on this project. Looks super cool! I hope my design helped.

Pete

alf45tar, 2019-08-30:
Thanks for sharing your experience.

marosell, 2019-09-14:
All, I included the above information in [my test board](https://github.com/alf45tar/PedalinoMini/issues/22), which I have sent to production and am waiting to receive.

Test board proposes MIDI Out via:
-  5V from the VIN pin
-  [M74VHC1GT08DTT1G](https://www.onsemi.com/pub/Collateral/MC74VHC1GT08-D.PDF)
> * +5v to MIDI through a 220R on Pin 4 on the MIDI DIN
> * The signal through the level shifter and a 220R on Pin 5 of DIN
> * Ground on Pin 2 of DIN

Will report back once tested.

prisme60, 2019-09-18:
Correction of the proposal in order to take into account of the MIDI out inverted connector pin 4/5 detected in #30 .

[https://easyeda.com/christian.foucher/pedalinomini-3v3-with-6n138-and-bss138-midi-out-pin4-5-now-ok](https://easyeda.com/christian.foucher/pedalinomini-3v3-with-6n138-and-bss138-midi-out-pin4-5-now-ok)

[image]

marosell, 2019-09-18:
> Correction of the proposal in order to take into account of the MIDI out inverted connector pin 4/5 detected in #30 .
> 
> https://easyeda.com/christian.foucher/pedalinomini-3v3-with-6n138-and-bss138-midi-out-pin4-5-now-ok
> 
> [image]

You sure the BSS138 will trigger on 3.3v? I didn't have a problem sourcing the M74VHC1GT08DTT1G, available at both Mouser and LSCS without a problem. But up to you on whichever you want to use.

prisme60, 2019-09-20:
BSS138 will adapt 3.3V to 5V without trigger effect. For example :
- 0V should give you (0/3.3)*5 ≈ 0V
- 1V should give you (1/3.3)*5 ≈ 1.51V
- 2V should give you (2/3.3)*5 ≈ 3.02V
- 3V should give you (3/3.3)*5 ≈ 4.53V
- 3.3V should give you (3.3/3.3)*5 ≈ 5V

It is a well known bi-directional level shifter, like I explain in one of my first message :  [Sparkfun](https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all), [Adafruit](https://www.adafruit.com/product/757).
Sourcing M74VHC1GT08DTT1G at Mouser costs 20$ of handling fee / shipping cost.
Sourcing M74VHC1GT08DTT1G at LCSC costs 7$ of handling fee / shipping cost.
The reason that I choose this component is :
- Delivery cost is very low.
- I have already this component (bought on [aliexpress](https://www.aliexpress.com/wholesale?catId=0&SearchText=BSS138&switch_new_app=y)).
**BUT I HAVE NOT STILL TESTED IT.**

I think I should also proposed the alternative of M74VHC1GT08DTT1G on the schema.

Remark : If it works well with BSS138, I won't buy M74VHC1GT08DTT1G, sorry I am "stingy".

jwyse, 2019-11-08:
This circuit works for me, using only 3.3v tapped from the ESP32, and no logic level converter.

[image]

This is essentially the same as the accepted answer [here](https://electronics.stackexchange.com/questions/165255/midi-in-many-circuits), except that the 6N138 can run off a 3.3v supply, and I didn't include the [power filtering?] capacitor on my breadboard.

As prisme60 mentions above, the 10K pull-down resistor on pin 7 makes the output signal rise noticeably faster.  The 1K pull-up resistor on pin 6 keeps the output between 0-3.3v -- without it, the output floats around 1.8v and the ESP32/PedalinoMini registers the noise as MIDI input.

The blue channel on the bottom is taken straight from the DIN connector, which is connected to my audio interface (and uses the 5v MIDI standard).  The yellow channel on top is measuring the output from the optocoupler, running at 3.3v.  (The channels are scaled differently, so the 5v signal is displayed smaller than the 3.3v signal.)

[image]

romakrau, 2019-11-29:
Hi Guys,
I found some different information concerning MIDI Spec 2014 (see enclosed picture). On MIDIin side is a current sink needed. IMHO no Level shifter nessesary. #
[image]

Greetings Roman

alf45tar, 2019-12-01:
Using Vtx = 3.3V and Ra=33 ohm and Rc=10 ohm resistors is in the standard since 2014 but according to Psychlist1972 (see above post) is not working an all devices.

athlonwai, 2020-03-03:
I've test MIDI OUT with 33ohm and 10ohm circuit with my USB Midi Interface(CME WIDI X8) as well as my Kemper, it does work normally.