alf45tar/PedalinoMini #14: FEATURE: Could this be modified to send SYSEX events?

n9yty, 2019-08-24:
Really looking for a way to control an FR-4x/FR-8x V-Accordion. The Roland FC-300 can be used, but it is $300. I have an FCB-1010, but it doesn't send SYSEX which is apparrently what the V-Accordion is expecting (you put the FC-300 in SYS mode).

This looked like it would be possible to retrofit inside an FCB-1010? :)  Anyway, a great project even if it can't do this.

alf45tar, 2019-08-24:
Are you sure you need SYSEX?
According to FR-8x MIDI implementation (https://static.roland.com/assets/media/pdf/FR-8_MIDI_Imp_e1.pdf) SYSEX in not supported.

n9yty, 2019-08-24:
When using the FC-300 for this purpose, to control certain features, it simply presents SYSEX in some fashion, in the accordion settings you only get to set pedal 1 to 9. It requires you put the FC-300 in SYS mode, so it isn't CC or PC messages. I had someone very knowledgeable about this (they set up the MIDI Solutions box to modify CC/PC to SYSEX) confirm that for me, which is why the FCB-1010 cannot work in this way for the accordion as it cannot send SYSEX.

If you look at the FR-4x/8x manual to see how to set this up, it tells you to put the FC-300 in SYS mode. Looking at the FC-300 manual it says that in SYS mode it seems SYSEX but I could quite get through enough of it to know if it says exactly what is sent.

Anyway, thanks for the consideration.

alf45tar, 2019-08-24:
You are right. I found the SYSEX MIDI implementation of FC-300.
It looks like that with FC-300 you don't need to configure the FR-4x/8x. But I don't think there is something that cannot do without SYSEX.
FR-4x/8x can be piloted also with CC/PC codes.

I am considering if add FC-300 SYSEX emulation to PedalinoMini.

n9yty, 2019-08-24:
Actually, in the FR-4x manual, I think it was page 80, you see that you simply can assign functions to pedals one through nine, but there is no configuration about how those pedals are defined. Must be the FC-300 in SYS mode sends out some known sequence when the pedals are pushed and the FR-4x is looking for those and only those.

While you can use CC and PC codes for their intended functions for note and program change, you cannot control these special functions using them according to anything I have seen.

underwoodblog, 2019-08-25:
MIDI-OX is a nice tool to try what message can be send and see what is send, when you have MIDI-Ports on your computer. http://www.midiox.com/