alf45tar/PedalinoMini #17: Display Issue with ESP32-WROOM-32U

marosell, 2019-09-01:
I built and uploaded to my ESP32-WROOM-32U (ESP32-DevKitC-32U) and it successfully uploaded. However, the screen shows static that slowly moves up a few lines at a time. I can confirm that the board briefly broadcasted a wifi name similar to "Pedalino", however, I pretty quickly used ESPTouch to provide the board with my wifi password.

[image]

I can now confirm by looking at my router that the board is connected to my wifi (device ID shows "ESP_09B3CC" is connected, but I am unable to navigate to ESP_09B3CC.local in my web browser, or the ip address to pull up the web user interface.

Any idea how I can fix the screen, or how I can find the web address without the screen?

underwoodblog, 2019-09-01:
Do you power it via the microusbcable? Try to change it with a quality one or power it from a PSU.

marosell, 2019-09-02:
> Do you power it via the microusbcable? Try to change it with a quality one or power it from a PSU.

I do power via USB micro cable, from computer. I tried different cables and different usb ports and it did not solve the issue. The issue was that I purchased a SH1106 OLED display and not a SSD1306 display, the latter is what is set as default in the build unless you use the TTGO_T_EIGHT board (which is unavailable).

Changing some of the if/else statements in the DisplayOLED.H file will solve the issue if using an SH1106 OLED display, however, it seems like there could be a way to either detect the display or make it easier to select the display before uploading to the board?

alf45tar, 2019-09-02:
If there is a way to auto-detect the display I will implement it. Otherwise I will move to SH1106 for both version and let user select SH1306 via compile flag.

alf45tar, 2019-09-18:
Added compile flags. Check platformio.ini

closed 2019-09-18 by alf45tar