alf45tar/PedalinoMini #18: Latch Switching Not Working

marosell, 2019-09-02:
I have successfully built and uploaded the esp32doit-devkit-v1-wifi build onto the ESP32-WROOM-32U (DevKit C) card. I have Momentary switches working correctly to execute single-press MIDI commands. However, I can't correctly implement Latch switching (is this the same as toggle?) with my switches. I assumed I would be able to select "Latch" from the Pedal Mode drop-down and the switch would alternate between MIDI Value 1 and 2 on the Bank page, but it doesn't seem to work that way.

Can you explain how this is supposed to function, and does it use MIDI Value 3 somehow? Do I have the circuit connected correctly? (I have the buttons connecting the D-ports to ground).

[image]

alf45tar, 2019-09-02:
A momentary pedal sends CC or PC on press. Nothing happens on release.
A latch pedal sends CC or PC on press and on release.

There are some exceptions. For example a momentary pedal sends a NOTE ON on press and a NOTE OFF on release.

Midi Value 2 and Midi Value 3 has a different usage. For example

marosell, 2019-09-02:
For switches between D-port and GND (for instance, I'm building an enclosure of only 6 momentary switches) do I still need to go back and add pull-ups to the A-pins? Or can we turn on the internal pull-ups for those pins in the code?

From the circuit diagram, I'm actually unclear as to why both a D-pin and an A-pin+PullUpResistor are required for each switch to ground; is the A-pin for the auto-sensing capability which turns the pedal on or off? (Forgive my naivety).

alf45tar, 2019-09-02:
It is not required to connect a pull-up resistor to A-pins using only Momentary1 pedal. GPIOs 34 to 39 are GPIs – input only pins. These pins don’t have internal pull-ups or pull-down resistors.

If you do not select the Analog pedal mode you can leave A-pins floating without the pull-up resistor. The A-pin is grounded in Momentary 1 schematic because is equivalent to inserting a TS jack.

alf45tar, 2019-09-02:
ERRATA CORRIGE

A latch pedal sends MIDI Code (CC or PC) on press and MIDI Value 2 on release.

marosell, 2019-09-03:
Out of curiosity, is there a reason why a TRS port switch cannot be used with Momentary1 (D and GND pins), by just grounding the ring with the sleeve together?

alf45tar, 2019-09-04:
I think I didn't understand well you question.

PedalinoMini TRS ports can be connected to a TRS jack or to a TS jack. Using a TS jack, ring (A-pin) and sleeve (GND) are connected together. This is not a problem for the A-pin because there is a pull-up resistor and a Momentary 1 is a switch between tip (D-pin) and sleeve (GND).

marosell, 2019-09-04:
I think I understand now. The circuit diagram on sheet 2/2 only shows TRS connectors on the left, including opposite of the first example, the Momentary1 (which of course can be used for Momentary1).

What is not clear is that a TRS connector is not _necessary_ for Momentary1. Only a TS jack is required (not shown), making the the resistor unnecessary for the A-pin. However, once a TS jack is used, it can **ONLY** ever be used for Momentary1 switching (and maybe Latch?) and never used for any of the other types of switches.

alf45tar, 2019-09-04:
I confirm:

- TRS jack is not necessary for Momentary1
- if TS jack is used do not connect A-pin
- once a TS jack is used, it can ONLY ever be used for Momentary1 switching Latch1

closed 2019-09-05 by marosell