alf45tar/PedalinoMini #19: Switch wireless networks without reset

marosell, 2019-09-02:
What's the proposed method for switching to a different wireless network without holding the boot button down until factory reset?

Use case: switching from wireless network at home for practice to wireless network at music venue, but don't want to lose saved settings.

alf45tar, 2019-09-02:
When you are at the music venue your home wireless network will not available then you can associate wifi credentials via SmarConfig.
Right now PedalinoMini can store only one wifi configuration. You have to repeat the wifi configuration when you are back to home again.

A possible improvement could be to link the wifi credentials to a profile or better to save a list of wifi credentials and rotating between them.

marosell, 2019-09-02:
Got it.

closed 2019-09-02 by marosell