alf45tar/PedalinoMini #20: Bank Feature Requests

marosell, 2019-09-03:
Was brainstorming this morning and came up with a few feature requests that others may find useful related to banks. Let me know if these should be separate issues:

1. Bank Switch Function: It appears that when pedal is in Bank+ or Bank- mode, you lose long press or double press functionality. Would love to have one "bank" button that can + on short press and - on long press. That way, only one button needs to be dedicated to bank navigation. Let me know if I'm doing it wrong.

2. Bank Disable: Allow user to disable 'n' banks after bank 1. That way, the Bank+ function loops back around to bank 1 with less presses if not utilizing all banks. When only using a few, #1 above isn't as important.

3. Bank Numbering: Suggest numbering banks 1,2,3,4,5,6,7,8,9,0, or 0,1,2,3,4,5,6,7,8,9. This allows ten banks but utilizes a one-digit number scheme. Suitable for 7-segment display (next bullet...)

4. 7-Segment Display: Allow bank information to be displayed via 7-segment display instead of or in addition to OLED screen. This allows a much bigger indicator on a pedal from standing height ~6 feet when playing an instrument.

marosell, 2019-09-04:
So I attempted to source a 7-segment display that could be driven via serial connection, and I couldn't find anything smaller than a 4-digit 7-segment display. While you can show more information on this, it defeats the idea of using a very small display with a very large font. That, and the power requirements are probably the same, if not more, than the small OLED screen.

I think number 4 above could be solved by showing the Bank information larger on the screen. Maybe "PedalinoMini" could be displayed on boot-up, but be replaced with the Bank information with a larger font in the information rotation. And flash the screen when changing banks, similar to how the switch information flashes on the screen when pressing footswitches.

alf45tar, 2019-09-04:
I was thinking the same thing:display bank on the existing OLED with a big font.

alf45tar, 2019-09-04:
Did you already try the TapDance mode?

alf45tar, 2019-09-08:
In the last version point 3 and 4 are solved with big fonts 
In "live performance" mode the current profile and the current bank is always displayed.
You can return to previous scrolling mode pressing BOOT button on DoIT and CENTER on TTGO.

alf45tar, 2019-09-21:
Point 2 implemented. Go to Pedals and use Min and Max. Bank+/Bank+ will move current bank between Min and Max value.

bobbyzero, 2019-10-22:
Now that #39 - momentary 2 pedal display has been fixed, how about an option to change bank by +/-2 with a single press of the button? 

That way each odd/even bank pair would be one 'virtual' bank and switching between them would require less activity.

mknerr, 2019-10-22:
I like that idea, and would only add that bank +3 would be great as well for those using momentary 3 pedals

alf45tar, 2019-10-23:
Thanks for the idea. I will add Bank+2 and Bank+3 feature. Of course Bank-2 and Bank-3 too.

alf45tar, 2019-10-23:
Did it.

alf45tar, 2019-10-25:
Point 1 of the first message implemented.
Double press or long press invert the pedal function.

alf45tar, 2019-11-07:
Can we close it?

bobbyzero, 2019-11-10:
I'm happy for the part I raised to be closed

closed 2019-11-11 by marosell