alf45tar/PedalinoMini #21: Minimum PCB Design

marosell, 2019-09-05:
I used Fritzing to put together a quick mockup of a bare minimum PCB design. Here were my goals with this design:

- Minimize board size, but allow board to be soldered by hand
- Allow screen, USB Midi, and physical midi port to be added via break-out boards.
- ESP32 board should be removable.
- Make connections for switches/jacks easier to identify and organize

To accomplish this, the ESP32 attaches to female headers on the board, and THT resistors are up underneath and out of the way. All other connections are male headers which can be connected via jumper wire. This allows hobbyist to only use the items they need, and/or expand or replace functionality later. Thoughts appreciated.

[PedalinoMini_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3577481/PedalinoMini_pcb.pdf)
[PedalinoMini_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3577483/PedalinoMini_etch_copper_top.pdf)
[PedalinoMini_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3577484/PedalinoMini_etch_copper_bottom.pdf)

marosell, 2019-09-05:
Quick revision for something a little more efficient, and makes the resistors easier to solder.

[PedalinoMini_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3577570/PedalinoMini_pcb.pdf)
[PedalinoMini_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3577578/PedalinoMini_etch_copper_top.pdf)
[PedalinoMini_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3577579/PedalinoMini_etch_copper_bottom.pdf)

alf45tar, 2019-09-05:
Cool!

marosell, 2019-09-05:
Alright, found some errors and made a few changes, including:

- Adjusted layout to make sure there's room for the antenna
- Added a few areas for 5V connectors and grounds (batteries options, etc)
- Added 5V pin for USB Midi Breakout
- Added 3.3V pin for Midi Breakout
- Screw holes
- Silkscreen Labels

Unless reviewed and other changes need to be made, I will probably close development of this layout and work on one for DevKitC (which is the board I have).

[PedalinoMini_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3580435/PedalinoMini_pcb.pdf)
[PedalinoMini_etch_silk_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3580436/PedalinoMini_etch_silk_top.pdf)
[PedalinoMini_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3580438/PedalinoMini_etch_copper_top.pdf)
[PedalinoMini_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3580439/PedalinoMini_etch_copper_bottom.pdf)

Let me know if you would like me to upload these somewhere or send the Fritzing file.

closed 2019-09-14 by marosell