alf45tar/PedalinoMini #22: Minimum PCB Design for DevKitC

marosell, 2019-09-05:
I plan to order this one, so if someone can review, I would much appreciate it. I will build and test once ordered. Minimum order is 3 boards; I'm happy to send a board to assist with development, just let me know where to send.

This is based on the ESP32-DevKitC-32U for a few reasons:
- Separates the screen from the board, allowing more placement options in an enclosure
- Provides antenna attachment, allowing for external antenna to be mounted to quality aluminum pedal enclosure

Link to board: https://aisler.net/p/PFQWIOVC

[PedalinoMini DevKitC_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3581605/PedalinoMini.DevKitC_pcb.pdf)
[PedalinoMini DevKitC_etch_silk_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3581607/PedalinoMini.DevKitC_etch_silk_top.pdf)
[PedalinoMini DevKitC_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3581608/PedalinoMini.DevKitC_etch_copper_top.pdf)
[PedalinoMini DevKitC_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3581612/PedalinoMini.DevKitC_etch_copper_bottom.pdf)

marosell, 2019-09-06:
Last revision for today. Updates include:

- Power options along the bottom of the board, including 9V power, switch header, and 9V/5V conversion header. If user is planning on a DC power connector on a breakout board, simply bypass the switch with a jumper. I recommend installing a bridge rectifier (Diotec B40R or similar) on the power breakout if using a power connector. 
- Power options for external midi module; 3.3V or 5v can be selected using a jumper.
- Unused pins from the board have pads for testing or future use.
- Grounded screw holes.
- General cleanup and simplification.

[PedalinoMini DevKitC_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3582422/PedalinoMini.DevKitC_pcb.pdf)
[PedalinoMini DevKitC_etch_silk_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3582423/PedalinoMini.DevKitC_etch_silk_top.pdf)
[PedalinoMini DevKitC_etch_silk_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3582424/PedalinoMini.DevKitC_etch_silk_bottom.pdf)
[PedalinoMini DevKitC_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3582425/PedalinoMini.DevKitC_etch_copper_top.pdf)
[PedalinoMini DevKitC_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3582426/PedalinoMini.DevKitC_etch_copper_bottom.pdf)

marosell, 2019-09-07:
Alright, we're at V. 1.0 on this board and it's ordered and off to production!

Final changes:

- Backtracked on grounded screw holes to avoid ground loops.
- Found room for the 9VDC bridge rectifier, so I added it. In lieu of the rectifier, 9V battery wires can be added to the +/- pins in this location.
- More cleanup.

[PedalinoMini DevKitC B_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3586243/PedalinoMini.DevKitC.B_pcb.pdf)
[PedalinoMini DevKitC B_etch_silk_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3586247/PedalinoMini.DevKitC.B_etch_silk_top.pdf)
[PedalinoMini DevKitC B_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3586248/PedalinoMini.DevKitC.B_etch_copper_top.pdf)
[PedalinoMini DevKitC B_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3586249/PedalinoMini.DevKitC.B_etch_copper_bottom.pdf)
[PedalinoMini DevKitC B_etch_silk_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3586250/PedalinoMini.DevKitC.B_etch_silk_bottom.pdf)

alf45tar, 2019-09-07:
Thanks Mike

marosell, 2019-09-12:
I actually found an issue with one of the pins, so I had to reorder. I found that for not much more room on the board, I could add Midi Out. So here is V. 2.0:

[PedalinoMini DevKitC C_pcb.pdf](https://github.com/alf45tar/PedalinoMini/files/3603863/PedalinoMini.DevKitC.C_pcb.pdf)
[PedalinoMini DevKitC C_etch_silk_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3603866/PedalinoMini.DevKitC.C_etch_silk_top.pdf)
[PedalinoMini DevKitC C_etch_copper_top.pdf](https://github.com/alf45tar/PedalinoMini/files/3603867/PedalinoMini.DevKitC.C_etch_copper_top.pdf)
[PedalinoMini DevKitC C_etch_copper_bottom.pdf](https://github.com/alf45tar/PedalinoMini/files/3603868/PedalinoMini.DevKitC.C_etch_copper_bottom.pdf)
[PedalinoMini DevKitC C_etch_silk_bottom_mirror.pdf](https://github.com/alf45tar/PedalinoMini/files/3603870/PedalinoMini.DevKitC.C_etch_silk_bottom_mirror.pdf)

mknerr, 2019-09-13:
Thanks for the very thorough response! And yes, I'd be interested in getting a board and the parts you have available. I think I have the rest anyway. Let me know how you would like to proceed. I think I setup an email in my profile.

marosell, 2019-09-14:
> Thanks for the very thorough response! And yes, I'd be interested in getting a board and the parts you have available. I think I have the rest anyway. Let me know how you would like to proceed. I think I setup an email in my profile.

I checked your profile and couldn't find it. I have added my email to my profile. I think it will be two weeks before the boards come in.

marosell, 2019-09-27:
Boards are in, I've built one and will be testing, and @mknerr will test one as well. Here are some pictures:

[image]
[image]
[image]

alf45tar, 2019-09-27:
Very nice and  good job.

marosell, 2019-09-29:
Was able to do some testing today and made a few short videos:

-Testing MIDI-out: https://youtu.be/BEKNbTDvX7c
-Testing IP-MIDI: https://youtu.be/dNtW2n366bU

athlonwai, 2020-03-03:
Hi Marosell,

What kind of buttons are you using in the video?

marosell, 2020-03-03:
> Hi Marosell,
> 
> What kind of buttons are you using in the video?

They were some cheap buttons that came with a breadboard kit. For an actual enclosure, I plan on using momentary foot switches from a pedal parts supplier.

mknerr, 2020-06-06:
Well, I know it’s been awhile, but thought I would give an update on my build with this PCB. It’s still a work in progress, and kind of a proof of concept at this point. She doesn’t know what she wants to be when she grows up. Lol.  Obviously I can’t use the MIDI DIN with this enclosure. Might change it out for a 3.5mm jack, or just “piggy-back” one on. 
Right now, I just have the two jacks for switches, but will definitely expand it. 
Big kudos, credit, and thanks go to @marosell for developing the PCB and, of course, to @alf45tar and other contributors for the software, making it possible for people with limited time and skills to make such a powerful controller. 

[image]
[image]
[image]
[image]

alf45tar, 2020-06-07:
Nice

closed 2020-11-16 by alf45tar

jisv48, 2020-11-18:
> I checked your profile and couldn't find it. I have added my email to my profile. I think it will be two weeks before the boards come in.

Hello Marosell, can we order the boards thru you or thru Aisler.net as you mentioned above. Thanks in advance

marosell, 2020-11-18:
Have at it!! https://aisler.net/p/GCLIIISP

I saw your PM and will respond. If anyone else wants a board, I can order a batch and send them out less than Aisler's min. order amount, but I think they have a good price on a pack of three, which will give you some to practice with.

A7F, 2021-01-28:
@marosell sooo I am planning to build a pedalino pedalboard but I was wondering if it would be doable something similar to the [midicommander](https://meloaudio.com/products/meloaudio-midi-commander-guitar-floor-multi-effects-portable-usb-midi-foot-controller-foot-switches) in the sense that it has all the I/O in the back side: the two midi I/O ports, usb midi port and two expansion ports for expression pedals. How do you deal with powering the whole thing? If I understand correctly your PCB, it still needs two power sources in case you use USB midi which is not super comfortable... I mean, the best thing would be to add some rechargeable batteries, right?
Also the PCB doesn't seem to support addressable rgb strips?

mknerr, 2021-01-28:
Apologies if I'm covering anything you already know. If you use USB MIDI, the micro will get power from the host. But, if you want to setup the pedalino as host, it would need power, I'd think you would still only need one supply. 

The board is setup to allow for a higher voltage input, like 9-12V, but converts it down to 5V with a buck converter, then sends it to the ESP32 board. You could tap tap off of that if you need to. You can also just use a 5V supply, or even just power with the micro USB. Just depends what your needs are. The power input circuitry is mainly for convenience if you are using on a pedalboard rig which typically has 9V available already. 

And yes, you could also use battery power.

A7F, 2021-01-28:
> Apologies if I'm covering anything you already know. If you use USB MIDI, the micro will get power from the host.

hey thanks for answering 😄 
I'm sorry, I think I'm getting confused... what about the ESP32? By looking at the PCB you assume the current coming from the Pro Micro (200mA I suppose?) is enough to power the whole thing (ESP32 as well) but what if I add let's say 10x WS2812B leds? Only those 10 leds draw 500mA...

davidin73, 2021-01-28:
Hi!!!! I'm (slowly😭) working on a @marosell' pcb (thanks Mike ,is a great pcb).
Here some enclosure' and a populated pcb' photos
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mknerr, 2021-01-28:
@A7F  no sorry, the host would power the pro micro only; I just meant you wouldn't have to power it separately. I guess I'm a little confused on what you meant by needing two power sources. 

But, regardless, one power supply, whether it's battery or mains, should do it, as long as it has the current you need for the number of LEDs you want to use.

FabrizioGrasso0289, 2021-03-23:
i apologize for this obvious question to those who are savvy in this. Where does the arduino Nano/Micro goes in this board?

basst22778, 2022-06-27:
@marosell i have problems finding the logic chip for midi out.
Do you think this chip from TI will match? 
https://www.reichelt.de/single-bus-buffer-gate-mit-3-state-ausgaengen-sot-23-5-sn-74lvc1g125dbv-p219307.html