alf45tar/PedalinoMini #25: Clarification on wiring schemes

jberry, 2019-09-08:
I noticed there is both Analog and Digital pins assign to a single momentary switch pedal when it has TRS.

So when using a TS-only pedal/regular footswitch, can we get away with only wiring it to either the Analog or the Digital input, and not both? If so, which is preferred, and should we pullup/pulldown the other pin?

alf45tar, 2019-09-08:
For TS jack only digital pin is required. PedaliniMini is using the internal pullup resistor for every digital pins.
Anlog pins can be be leave floating (not connected) if you use only Momentary 1 or Latch 1 pedal.

I also suggest to disable Autosensing in order to not automatically enable Analog.

jberry, 2019-09-08:
I've got auto-sensing turned off, pedal set to 'Momentary' Bank+ function, bank set to Bank Select+ MIDI, I can see the switch close via a multimeter (D25 makes connection to Ground), but neither the serial console or screen show anything. 

Bank  1
Pedal  1     BANK_PLUS    MOMENTARY1                 POLARITY+      Channel  1   Pin D25

Any idea what to check next/if there is another setting/code definition that may need changed?

jberry, 2019-09-08:
I've got this working in Single Press mode, but it seems I have to hold the switch for 1second for a bank change even without Long Press turned on.  Is that by design? If so, is there a way to tweak it to be shorter?

closed 2019-09-08 by jberry