alf45tar/PedalinoMini #26: Repeat Press Time not saving in Config interface

jberry, 2019-09-08:
Go into Config, set a Repeat Press Time other than the 0 it's defaulted to, click Save, and it's still at zero.

It doesn't appear to be an interface-only display issue, as I'm able to cause repeat presses shorter than the times I specified

alf45tar, 2019-09-09:
Are you using the latest code? If yes try to reset PedalinoMini to factory default.

jberry, 2019-09-09:
I want to setup the repeat for Bank Select +/-, so that limitation is alright.
The prior flash was an 'upgrade' (flash of new code over old code), based on code pulled 9/8.
I just used esptool and wiped my esp32 with erase_flash, flashed with the latest from master, and it still doesn't seem to save (value continues to be zero in Config)

marosell, 2019-09-09:
There are two different save buttons in the interface. Which did you use? Or did you try both?

jberry, 2019-09-09:
Just tried the ABC bank save button, it didn't even popup the Saved green banner. Refreshed to make sure that banner wasn't just missing, and still saved at 0.

Tried the Save button at the bottom of the page (that works to save all the other fields), and it pops the green Saved banner, but Repeat Press Time never saves/shows anything other than 0.

marosell, 2019-09-09:
What happens if you change multiple values at the same time and save. For instance, set a new long press value at the same time as you set a new repeat press value and hit save.

marosell, 2019-09-09:
Found it.

Line 2021 of WebConfigAsync.h:

`  if (request->arg("repeatpressttime").toInt() != repeatPressTime) {`

Should be:

`  if (request->arg("repeatpresstime").toInt() != repeatPressTime) {`

jberry, 2019-09-09:
Was just testing fixing that line and working on forking so I can submit a PR for it; if someone beats me to it, all yours

PR #28 (2019-09-09): Removing duplicate T from string
Fixes #26
closed 2019-09-09 by alf45tar