alf45tar/PedalinoMini #27: Switching profiles using ESP32 Dev Kit

Hans68, 2019-09-09:
With TTGO-T-Eight switching profiles is possible with the navigation right/ left button. Is there a possibility to connect a similar switch to one of the the ports of ESP32 Dev Kit to do the same? I suppose, the navigation right/ left button is also connected to one of the ports of ESP on TTGO.
Thanks and Cheers

alf45tar, 2019-09-09:
Unlikely there are not too much GPIOs available.
GPIO37 and GPIO38 are not available on ESP32 Dev Kit. You can try with GPIO16 and GPIO17 changing the code (Pedalino.h) according.

Hans68, 2019-09-09:
Thanks, but doesn't work. Changing 37->16 and 38->17 ends in a SW freezing , Pedalino Display remains without changing to Device Name etc. 
,

alf45tar, 2019-09-09:
Fixed the code to support RIGHT on GPIO16 and LEFT on GPIO17 on ESP32 DOIT DevKit.

Hans68, 2019-09-09:
Great Thanks! 
I   suppose 10k pull up to 3,3V and contact to ground for active switching !?

alf45tar, 2019-09-10:
It is using the internal pull-up resistor. Just a switch to GND is enough.

Hans68, 2019-09-10:
Works fine, thanks !
Display is showing profile and Bank number in full display size, good solution !
Is Midi Out and Midi In deleted through this modification (it is also using Tx2 and Rx2) ? It is not a problem for me, I am not using it, but could be a problem perhaps for someone else.

alf45tar, 2019-09-10:
Serial2 is re-mapped to GPIO 4 Tx and GPIO 15 Rx.

closed 2019-09-10 by alf45tar