alf45tar/PedalinoMini #29: Wifi connection

rigr, 2019-09-13:
Hello,
first of all I'd like to thank you alf45tar for this project. I love to read your code  - it's systematic and clear.
I'd like to propose a small addendum:
The SmartConfig sometimes does not work for me (or I need to search my smart phone). Then pedalino goes into AP mode. 
I'd like to see a possibility to enter the WiFi credentials manually, like shown in https://github.com/zhouhan0126/WIFIMANAGER-ESP32, so we would need a switch on the webinterface in AP mode, that would allow to show the AP connect- page.
Additionally I'd also like to be able to delete the Wifi credentials saved on the chip (another button to be clicked).
What do you think?
Thank you and best regards
Richard

alf45tar, 2019-09-14:
SmartConfig is very stable for me.
There are very seldom cases where SmartConfig timeout is reached before obtain the credentials. This is because the timeout is only 15 seconds. I suggest to increase the Smartconfig timeout to 30 seconds if you want to be sure. I prefers to reboot PedalinMini when it happens (1% of the cases). The second try is always work for me.

WIFIMANAGER is very good but SmartConfig is faster and easier.

Added a boot option for reset WiFi credentils.

Thanks for your suggestion

rigr, 2019-09-14:
Thank you alf45tar,
wow, you impress me, I very much appreciate your work. 
Tonight and/or towmorrow I'll have a look and try your new code.
Thank you and best regards
Richard

closed 2019-09-21 by alf45tar