alf45tar/PedalinoMini #30: MIDI Schematic wrong?

jberry, 2019-09-14:
Trying to get midi jacks working, and they arn't.  Looking at the original Pedalino wiring, and the MIDI standard via sweetwater, I'm thinking the schematic may be wrong.

The MIDI Out standard I found is:

O….. Pin 1 (No Connection)
O……… Pin 4 (+5V DC Voltage)
O………… Pin 2 (Ground/Common)
O……… Pin 5 (MIDI Data – Hot)
O….. Pin 3 (No Connection)

And looking at the original Pedalino project wiring picture, I see that setup.

Looking at the Pedalino schematic, it seems like pin 4 and pin 5 may be swapped.  PIN 5 is getting the +5v signal, and PIN 4 is getting the data from the ESP32.  Also, I don't see a ground on PIN 2.

Could you review and see if you agree?

jberry, 2019-09-14:
I followed https://learn.sparkfun.com/tutorials/midi-tutorial/hardware--electronic-implementation and flipped the pins as I thought were wrong above, and it did fix my issue.  So it does appear the pins on the schematic are wrong

alf45tar, 2019-09-16:
I think you are right. I will fix it shortly.

alf45tar, 2019-09-16:
Fixed

Thanks @jberry

closed 2019-09-16 by alf45tar

prisme60, 2019-09-18:
I see that you only change [Schematic_PedalinoMini_Sheet-1.svg](https://github.com/alf45tar/PedalinoMini/blob/master/images/Schematic_PedalinoMini_Sheet-1.svg) but you should change 
[Schematic_PedalinoMini_Sheet-0.svg](https://github.com/alf45tar/PedalinoMini/blob/master/images/Schematic_PedalinoMini_Sheet-0.svg) too.

alf45tar, 2019-09-18:
Deleted Sheet-0 because Arduino Uno is not reccomended anymore for USB MIDI.
Arduino Pro Micro is cheaper, much compact and don't need the MOCOLUFA firmware.