alf45tar/PedalinoMini #32: Build Error

marosell, 2019-09-25:
Did a new git clone this evening and received the following error when trying to build:
In file included from src/PedalinoMini.cpp:71:0:
```
src/WifiConnect.h: In function 'void WiFiEvent(system_event_id_t, system_event_info_t)':
src/WifiConnect.h:258:10: error: 'SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP' was not declared in this scope
     case SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP:
          ^
Compiling .pio/build/esp32doit-devkit-v1-wifi/libb2e/BLE/BLEDescriptor.cpp.o
*** [.pio/build/esp32doit-devkit-v1-wifi/src/PedalinoMini.cpp.o] Error 1
/Users/Mike/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(float&)':
/Users/Mike/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristic.cpp:674:16: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((float*)temp) = data32;
                ^
/Users/Mike/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(double&)':
/Users/Mike/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristic.cpp:680:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((double*)temp) = data64;
                 ^
```

My clone from Sept 2, 2019 still builds without this error. I'm building with esp32doit-devkit-v1-wifi with SSH1106WIRE build flag.

Any ideas?

alf45tar, 2019-09-26:
Update the ESP32 platform inside PlatformIO or delete the case SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP.

SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP is a new event introduced with the latest version.

marosell, 2019-09-27:
That did it

closed 2019-09-27 by marosell