alf45tar/PedalinoMini #33: SmartConfig Error

marosell, 2019-09-27:
Having an error with the latest build where I cannot provide WIFI credentials via the Esptouch app running on iPhone, where this was working stable a month ago.

Problem: When running Esptouch on an iPhone XR running latest 13.1, the PedalinoMini freezes at the end of SmartConfig as shown:
[image]

If rebooting and NOT running Esptouch, SmartConfig times out and continues through the boot cycle just fine.

FWIW, it is possible that Esptouch app may need to be updated for iOS 13.1, and this may not be an issue with this code specifically.

alf45tar, 2019-09-27:
I confirm ESPtouch v0.3.7.0 doesn't work on iOS 13.1.
It looks like it doen't recognize SSID (Wi-Fi) and BSSID (00:00:00:00:00:00).

marosell, 2019-09-27:
Confirmed. I opened up my iPad which I had not updated yet (still running 12.3.1) and was able to use Esptouch on the latest PedalinoMini build and communicate.

Feel free to close this issue, or leave it open as the title is informative for other folks (until Esptouch is fixed).

alf45tar, 2019-09-28:
Open an issue on official EsptouchForIOS repository https://github.com/EspressifApp/EsptouchForIOS/issues/40#issue-499740632

closed 2019-10-19 by alf45tar