alf45tar/PedalinoMini #34: Single Press Cannot Be Disabled

marosell, 2019-09-29:
I have a Momentary Pedal on Pedal#4 set as Control Change. I have enabled Long Press and disabled Single Press. Single Press still sends Control Change.

alf45tar, 2019-10-05:
Thank for your debugging.

I think the issue is a little bit different. By default single press should be very fast otherwise it is recognized as long press.

The issue is that double press and long press are enabled together. They cannot be disabled one by one.

I let you know when fixed.

alf45tar, 2019-10-05:
Fixed. 
Let me know if it is ok for you.

mknerr, 2019-10-07:
I seem to be having issues as well, and I am using the latest source code. I know I was able to setup long presses in previous versions, and I can't seem to figure it out now, but apparently it's because of the bug.

I see you've added settings for the button press times in the Options tab. Will a setting of "0" leave them at previously established defaults, or is the setting actually "0"? If the latter is the case, I'm assuming that could affect the behavior too. What would you recommend for those values?

alf45tar, 2019-10-08:
0 is zero. It is not the previous value.

Default values (after resetting to factory default) are:

Press Time              200
Double Press Time 400
Long Press Time     500
Repeat Press Time  500

You can change them in Options without resetting to factory default.

Cheers
alf45tar

mknerr, 2019-10-09:
Okay, thank you! I put in those time values in the Options tab, and single, double, and long presses seem to be working again. But to clarify, when I updated, those were all set to zero. So, obviously, if those were pre-populated with the defaults, that would be ideal. 
I am interested to see if that is @marosell 's issue as well.

marosell, 2019-10-17:
Not certain, but I believe the default values were empty last time I was working with it. I'll grab the latest and test it with these values if they are not the default and report back.

closed 2019-12-18 by alf45tar