alf45tar/PedalinoMini #35: Can't get ladder to work

jberry, 2019-10-09:
I have what I believe should be a working Ladder switch wired up, but the interface/serial console does not register any behavior when a switch is pressed in my ladder setup.

I've checked ohm across my Ladder setup, and it matches the schematic for tip/ring/sleeve.

I've checked the pin input, including the pullup resistor. 

I've switched the pin to a standard momentary input, and it works fine on that setting.

Can you point me what to debug/try next? I'm comfortable in VSCode (although my C is rusty) and starting to dig through to see what else I can I can poke.

jberry, 2019-10-10:
Here's the serial console output of the settings of the first four pedals.  I can change pedal 3 & 4 to MOMENTARY2, and they work fine with the same pedals plugged into 1 and 2; but when changed to Ladder and plugging in my ladder pedal, it registers absolutely nothing even though I can register changes through my multimeter. I noticed the Ladder pedals don't have a Digital pin?

> Pedal  1     MIDI         MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE   0   Channel  2   Pin D25 A36
> Pedal  2     MIDI         MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE  62   Channel  2   Pin D26 A39
> Pedal  3     MIDI         LADDER       PRESS_1       POLARITY+   CONTROL_CHANGE  64   Channel  2   Pin A34
> Pedal  4     MIDI         LADDER       PRESS_1       POLARITY+   CONTROL_CHANGE  65   Channel  2   Pin A35

alf45tar, 2019-10-11:
I suggest to stop your testing with ladder because it is not supported yet.
I will do my best to release ladder support soon.

alf45tar, 2019-10-26:
@jberry test of ladder can start.

jberry, 2019-10-27:
@alf45tar I did a pull, PlatformIO:Clean, then tried to build esp32doit-devkit-v1-wifi, and it's throwing the following. It looks like VSCode/PlatformIO self-updated, so not sure if that did something. 

> In file included from src\PedalinoMini.cpp:69:0:
> src\WebConfigAsync.h: In function 'void http_run()':
> src\WebConfigAsync.h:2390:15: error: 'class AsyncWebSocket' has no member named 'cleanupClients'
>      webSocket.cleanupClients();
>                ^
> In file included from src\PedalinoMini.cpp:71:0:
> src\WifiConnect.h: In function 'void WiFiEvent(system_event_id_t, system_event_info_t)':
> src\WifiConnect.h:279:10: error: 'SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP' was not declared in this scope
>      case SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP:
>           ^
> *** [.pio\build\esp32doit-devkit-v1-wifi\src\PedalinoMini.cpp.o] Error 1

jberry, 2019-10-27:
Ladder does not appear to be behaving any differently, as well as my 2 button switches (Momentary 2) no longer work on the latest code.

jberry, 2019-10-28:
@alf45tar Alrighty, I have a mostly-working test-bed.  I only have a 4-switch ladder at the moment, working on 3d printing my 5switch case to get a 5switch ladder wired up.

Probably saw some other messages in this issue, I deleted those; Had some PlatformIO issues (had to pio uninstall lib the Async web server so it force recompiled, not sure why; then had a wire come disconnected that I totally missed that caused some weird fun).

Are the Ladder switches meant to be Momentary or Latching? Or does there need to be 2 Ladder types, one for each? I have it wired with Momentary, so I notice that it's sending a signal on both press and release

alf45tar, 2019-10-28:
Only Momentary switch works with ladder.
If you press 2 or more switches only the first has a predefined behavior. The second will have an undefined behavior.
Ladder is a cheap trick. If you need independent switches a matrix should be used but 1 analog pin is not enough to support it.

jberry, 2019-11-05:
I have a 4 switch ladder and a 5 switch ladder wired up.  In both cases, button #4 does not work.

On the 5 switch ladder, switch 5 works fine.

alf45tar, 2019-11-07:
Can you measure the volts on analog pin when switch 4 is pressed?
You should measure a value between 0.8V and 1.8V.

jberry, 2019-11-25:
Finally tracked this down to my vendor sent the wrong 1k resistors; They were small enough values that switch5 was actually registering as switch 4, and I really had no switch 5.

closed 2019-11-25 by jberry