alf45tar/PedalinoMini #37: Proper behavior of momentary 3 pedal

mknerr, 2019-10-11:
I’ve been testing different types of switches and pedals and I have not been able to get my momentary 3 type pedal to work correctly, but I’m not entirely sure how it is supposed to function. My assumption is that I would set the corresponding pedal input to momentary 3, then configure the desired behavior in the banks tab. Presumably, then each switch would trigger MIDI value 1, 2, and 3, respectively. ??
If that’s correct, I’ll probably focus on the hardware and connections, but I’m using a TC electronic switch, and @marosell ‘s PCB, which should limit issues there. 
At best, I’m getting two values, and the response is pretty erratic. Sometimes I’ll get two values on one press.

alf45tar, 2019-10-11:
Switch 1 use current bank
Switch 2 use current bank + 1
Switch 3 use current bank + 2

mknerr, 2019-10-11:
Thank you!! It's working perfectly, now that I know what it supposed be doing. And, the functionality is better than I had hoped.

closed 2019-10-11 by mknerr