alf45tar/PedalinoMini #38: Bank Change pedal always uses long press delay

bobbyzero, 2019-10-13:
I have two buttons set up to control bank changes, connected to pedal 1 D/A pins:
- Pedal 1 - Momentary 2 - Bank+
- only Single Press active, Min 1 / Max 10

When long press delay is set to default 500, these buttons need to be held down for that time to trigger a bank change.

Reducing long press delay to 0 makes the change instant, raising it to 5000ms means a 5 second press is needed, so it's definitely reading this value.

bobbyzero, 2019-10-13:
Looking at Controller.h I think that it's only setting the values correctly for pedals that are marked as MIDI

`if (pedals[i].function == PED_MIDI) {...}
else
            {
              pedals[i].footSwitch[p]->setDoublePressTime(doublePressTime);
              pedals[i].footSwitch[p]->setLongPressTime(longPressTime);
              pedals[i].footSwitch[p]->setRepeatTime(repeatPressTime);
              pedals[i].footSwitch[p]->enableRepeatResult(true);
            }`

alf45tar, 2019-10-14:
What happen if you press and release button very quickly?

bobbyzero, 2019-10-14:
When long press is set, the buttons don't do anything unless I hold them down for that time.
A quick press, or repeated quick presses don't have any result

alf45tar, 2019-10-19:
> When long press delay is set to default 500, these buttons need to be held down for that time to 
 trigger a bank change.

>Reducing long press delay to 0 makes the change instant, raising it to 5000ms means a 5 second press is needed, so it's definitely reading this value.

I did some additional testing and what you are saying is correct but the usual way to trigger an event is to press and release the button. The bank changes on release.

If you need to trigger on press set long press to 0.

alf45tar, 2019-10-20:
@bobbyzero try now.
It should work as expected. It is now more reactive to button press and repeat works better.

bobbyzero, 2019-10-21:
Seems to be working great now, I'm not sure I ever had repeat working before. Thanks!

closed 2019-10-21 by bobbyzero