alf45tar/PedalinoMini #40: "AP mode failed" since commit 5197999

jwyse, 2019-10-27:
Fresh install on ESP32.  No previous configuration.  VSCode, PlatformIO, boards, etc. are all up-to-date.  Build succeeds but I get the below error at runtime.  I tried a few different commits and found that the issue started with commit 517999 from Oct 19.

latest commit, **7e9f1d8**... (Oct 26)

> [...] // serial monitor during bootup:
> DIN MIDI started
> Skipped connection to last AP and/or SmartConfig/WPS setup
> SYSTEM_EVENT_WIFI_READY
> **### Starting AP [] with password [] // <- I added this DPRINT for debugging
> [E][WiFiAP.cpp:103] softAP(): SSID missing!
> AP mode failed**
> Internal Total Heap 323120, Internal Free Heap 242560
> SYSTEM_EVENT_AP_START
> Loading profile ...
> Reading NVS Profile [E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
> Pedal autosensing...

Relevant code from `WifiConnect.h`, with an added DPRINT line:

```
    DPRINT("### Starting AP [%s] with password [%s]\n",
                ssidSoftAP.c_str(), passwordSoftAP.c_str());
    if (WiFi.softAP(ssidSoftAP.c_str(), passwordSoftAP.c_str())) {
        DPRINT("AP %s started with password %s\n", ssidSoftAP.c_str(), passwordSoftAP.c_str());
        // [...]
        start_services();
    }
    else
        DPRINT("AP mode failed\n");
```

This issue seems to have been introduced in commit 5197999 when the `ssidSoftAP` variable was introduced alongside, and sort of replacing, `wifiSSID`.  At this point within WifiConnect.h, ssidSoftAP and passwordSoftAP are null or empty.  It appears that there are no local variables by those names.

The previous commit, 3710def from Oct 12, **does not** have this issue, although it does seem to have an issue with mDNS that I have not tried to chase down.

> SYSTEM_EVENT_WIFI_READY
> **AP Pedalino-3C40A24 started with password
> SYSTEM_EVENT_AP_STOP
> AP SSID     : Pedalino-3C40A24
> AP PSK      :
> AP MAC      : 24:0A:C4:03:A6:D5
> SYSTEM_EVENT_AP_START
> SYSTEM_EVENT_AP_START
> AP IP       : 192.168.4.1
> Channel     : 1
> Connect to Pedalino-3C40A24 wireless network with password**
> [I][ArduinoOTA.cpp:356] end(): OTA server stopped.
> [E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
> [I][ArduinoOTA.cpp:130] begin(): OTA server at: esp32-240ac403a6d4.local:3232

alf45tar, 2019-10-28:
Did you try a reset PedalinoMini to factory default?
The settings saved into EEPROM are changed in commit 5197999.

jwyse, 2019-10-28:
Resetting to defaults did get past this issue, but revealed two more.  I'm about to submit a pull request that should address the first one, and prevent the second one.

I think you should be able to reproduce my issue (and upcoming pull request) by factory-resetting with the latest code and boards/libraries in PlatformIO.

1. The default AP password is the chip ID, which is 6 hexadecimal characters.  The ESP32 wifi library now requires at least 8 characters for the password, so PedalinoMini failed with the following error:

```
WPS timeout
SYSTEM_EVENT_AP_START
[E][WiFiAP.cpp:109] softAP(): passphrase too short!
AP mode failed
```

[WiFiAP](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiAP.cpp#L107) now requires 8 characters for the passphrase.
```
    if(passphrase && (strlen(passphrase) > 0 && strlen(passphrase) < 8)) {
        // fail passphrase too short
        log_e("passphrase too short!");
        return false;
    }
```

2. Unhandled exception during first boot after resetting defaults, AND any time a button was pressed. 
 This unhandled exception crashed/rebooted the ESP32.  Caused by an inconsistent state where the `wifiEnabled` variable is true, but the wifi connection failed (not configured as a client, and AP failed), and it tries to send an ipMIDI message with no connection.

Serial output from bootup, with no buttons pressed or held after triggering the factory default:

```
Hostname: 3C40A24
Internal Total Heap 324672, Internal Free Heap 298108
PSRAM Total Heap 0, PSRAM Free Heap 0
Reading NVS Global ... done
####################################################################################################################################################
Reset EEPROM to factory default
Updating NVS ...
done
[NVS][Global][Device Name]: 3C40A24
Updating NVS ...
done
[NVS][Global][Boot Mode]: 1
Updating NVS ... done
[NVS][Global][STA SSID]:     
[NVS][Global][STA Password]: 
Updating NVS ... done
[NVS][Global][AP SSID]:     Pedalino-3C40A24
[NVS][Global][AP Password]: 3C40A24
Updating NVS ... done
[NVS][Global][Bootstrap Theme]: bootstrap
Updating NVS ... done
[NVS][Global][Current Profile]: 0
Updating NVS ... done
[NVS][Global[Tap Dance Mode]: 0
Updating NVS ... done
[NVS][Global[Bank Switch]: 0
Updating NVS ... done
[NVS][Global[Single Time]: 200
[NVS][Global[Double Time]: 400
[NVS][Global[Long   Time]: 500
[NVS][Global[Repeat Time]: 500
Updating NVS ... done
[NVS][Global[Blynk Cloud]: 0
Updating NVS ... done
[NVS][Global][Blynk Token]: 
Updating NVS Profile A ... done
Updating NVS Profile B ... done
Updating NVS Profile C ... done
Reading NVS Global ... done
USB MIDI started
DIN MIDI started
SYSTEM_EVENT_WIFI_READY
[... 71 lines cut ...]
  ipMIDI.write(midiPacket, 1);
  ipMIDI.endPacket();
}
```

I solved this by changing the default passphrase to repeat the chip ID.  It is now 12 characters instead of 6, and still predictable based on the chip ID which is part of the default SSID.
```
void eeprom_update_ap_wifi_credentials(
    String ssid = String("Pedalino-") + getChipId(),
    String pass = getChipId() + getChipId())
```
It now boots and handles button presses without crashing/rebooting.
```
AP Pedalino-3C40A24 started with password 3C40A243C40A24
SYSTEM_EVENT_AP_START
AP SSID     : Pedalino-3C40A24
AP PSK      : 3C40A243C40A24
AP MAC      : 24:0A:C4:03:A6:D5
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
AP IP       : 192.168.4.1
Channel     : 1
Connect to Pedalino-3C40A24 wireless network with password 3C40A243C40A24
```

alf45tar, 2019-10-28:
@jwyse thank you very much for your debugging and very precise report.

The bug is in the getChipId function. It should return 8 hex digits from MAC address (the low 4 bytes) converted to a String. Due to a bug in my implementation it return only 7 digit when the first is zero like in your case 03C40A24.

alf45tar, 2019-10-28:
@jwyse How do you decode backtrace in Platformio?

jwyse, 2019-10-28:
You don't.  :(
There are some workarounds referenced in [this open issue](https://github.com/platformio/platform-espressif8266/issues/31).

I used [me-no-dev's EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) in the Arduino IDE.  It's just a wrapper within Arduino IDE, so you could use a command line version instead.

If you do install the tool in the Arduino IDE, you can use it without actually loading/building/flashing the code there.  Just launch the tool, point it to the ELF file in <PedalinoMini>\.pio\build\esp32dev\firmware.elf, and then paste the backtrace string.

jwyse, 2019-10-29:
Your latest commit a few hours ago solved this problem.  This issue can be closed.  Thanks!

closed 2019-10-29 by alf45tar