alf45tar/PedalinoMini #44: Bluetooth not advertising

jberry, 2019-11-25:
I've booted my pedalinomini into Bluetooth only mode, and trying to pair with it.  I don't see any new available pair-able devices as I scan, but I see a few errors in the serial monitor.  Any suggestions?

[bluetooth-no-advertise.txt](https://github.com/alf45tar/PedalinoMini/files/3887682/bluetooth-no-advertise.txt)

[D][BLEDevice.cpp:556] getAdvertising(): get ad[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 4]
Loading profile ...
Reading NVS Profile A ... [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 1 0x01
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
Bank  1
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
ENTARY2   [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 6 0x06
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
   Pin D25[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]

alf45tar, 2019-11-26:
The issue is not linked to BLE but to NVS.

Reading NVS Global ... [E][Preferences.cpp:437] getString(): nvs_get_str len fail: Device Name NOT_FOUND
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: AP SSID NOT_FOUND
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: AP Password NOT_FOUND
[V][Preferences.cpp:306] getUChar(): nvs_get_u8 fail: Current Profile NOT_FOUND
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: Single Time NOT_FOUND
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: Double Time NOT_FOUND
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: Long   Time NOT_FOUND
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: Repeat Time NOT_FOUND
done

Try to reset to factory default.

jberry, 2019-11-26:
I wiped it with esptool erase_flash, flashed Pedalino from a fresh git clone, did a Factory Default boot, connected it to my AP, set the device name and the pedal timing settings, rebooted a few times, ensured the web interface held my settings, rebooted into BLE only mode, and still no advertising that I can see from 2 different devices.

[freshboot.txt](https://github.com/alf45tar/PedalinoMini/files/3893510/freshboot.txt)

Reading NVS Global ... [E][Preferences.cpp:437] getString(): nvs_get_str len fail: AP SSID NOT_FOUND
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: AP Password NOT_FOUND
[V][Preferences.cpp:306] getUChar(): nvs_get_u8 fail: Current Profile NOT_FOUND
done

alf45tar, 2019-12-01:
Preferences.cpp errors should disapper after a Factory Default boot.
I cannot reproduce the issue.

jberry, 2020-01-05:
I got a second/different ESP dev kit off amazon, and had the same behavior.   I loaded up bluetooth on my laptop since it scans faster and more frequent than my phone/tablet, and it every now and then saw the pedalino bluetooth signal, but it'd quickly disappear.  The VScode serial monitor showed the same behaior as the first.

If I did get a chance to tell it to pair before it disappeared, it would once but it'd quickly disconnect and never self-reconnect. The logs showed many of the same dumpGapEvent errors, but nothing new/useful it seemed.

I've since abandoned my attempt at building a pedalino and bought commercial.  I'm happy to provide logs from that last attempt/pair/disconnect behavior if you'd like, or you can just close this unable to reproduce since it seems only I had this issue/no one else has commented 'me too'.

alf45tar, 2020-01-05:
Just to be sure we are on the same page.
Which application are you using for connect? BLE is not Bluetooth. You cannot connect to a BLE-MIDI device without an app.  For iOS I am using GarageBand or midimittr.
On Windows you can use MIDIberry but again your laptop must support BLE. Bluetooth is not enough.
Never tried Android because it is a mess. Too many different devices.

Cheers
alf45tar

closed 2020-01-11 by alf45tar