alf45tar/PedalinoMini #45: Pedalino Mini as Wlan-Midi bridge

romakrau, 2019-11-25:
Hi Alf45tar, Thanks for this great tool. It works out of the box for a small budget.
It is possible to use this ESP32-Dev kit also as MIDI bridge? I'm mean that on the instrument side we can use the DIN-MIDIin and MIDIout and receive the MIDI data via Wlan or Wifi by another ESP32-DEV with DIN MIDIin / MIDI out.
Greetings
Roman

alf45tar, 2019-11-26:
Yes, it is possible via IPMIDI because it is a broadcast protocol.
Using rtpMIDI you need a computer (session listener) because PedalinoMini is a session initiator only.

romakrau, 2019-11-26:
I tried to install rtpMIDI from tobias erichsen. I can see the pedalino but i have to prepare the midi connector for testing. Thanks for your quick response.

closed 2019-11-26 by romakrau