alf45tar/PedalinoMini #47: On TTGP T8 no display(black lit screen) however on web shows remote display ok

davisaminaden, 2019-12-10:


alf45tar, 2019-12-13:
TTGO T8 has no built-in display.

TTGO T-Eight has one but if you own the old model change https://github.com/alf45tar/PedalinoMini/blob/64a0239e5ac63bf6be133eadb2847beace6cc99e/src/DisplayOLED.h#L17 and https://github.com/alf45tar/PedalinoMini/blob/64a0239e5ac63bf6be133eadb2847beace6cc99e/src/DisplayOLED.h#L18

because SDA is 5 and SDL is 4.

davisaminaden, 2019-12-18:
Change it to what value?
You mean remove the '#' ?

davisaminaden, 2019-12-18:
I have this board
https://joom.app.link/sDcjPBDUw2

alf45tar, 2019-12-18:
Nice board but the IPS display (ST7789V) is not supported (yet) by PedalinoMini.

davisaminaden, 2019-12-19:
So I cannot display anything on this you mean right?
Also another thing I noticed
if I go to the main web page and click on any other link , the board crashes and resets
is this a known issue?

alf45tar, 2019-12-20:
Yes, you cannot display anything on build-in display. You could attach an external display.

Yes, it is a know issue. Board with 4MB has no enough memory to support Wifi and bluetooth at the same time. I suggest to disable use Wifi or bluetooth. You can switch between modes at boot time.

closed 2019-12-25 by alf45tar