alf45tar/PedalinoMini #49: Request for status Led

ciapamusche, 2019-12-15:
Not an Issue. I plan to built PedalinoMini but i like to have the status led/bypass on my switches. so i know ever if the effect is active or not. Like a normal pedal board. 

The top must  be have a midi feedback from software for that. So if in my case from Amplitude i switch off the effect, the Led switch off. 

Thank you for sharing your project

alf45tar, 2019-12-16:
I didn't add the status led yet because there are no more GPIOs available. I had to add some hardware to expand GPIO.

I let you know what we can do.

alf45tar, 2019-12-16:
@ciapamusche are you sure that Amplitube can send MIDI messages or just receive them?

alf45tar, 2019-12-16:
@ciapamusche I confirm there is no MIDI output on Amplitube. Only MIDI input.

> When running as a plug-in, AmpliTube 4 will be able to receive MIDI from the host. Refer to your host sequencer or DAW on how to set it up so that it can send MIDI to audio plug-ins. When running as a Standalone, AmpliTube 4 needs to be configured to know from which MIDI port it should read MIDI data from.

mlsomers, 2019-12-16:
You could use a "shift register" to multiply GPIO outputs. You need two GPIO pins, one for the state and one for what's called a "clock" but it basically pushes the state to the next pin on each "clock cycle". This is how segment-LED display's are usually driven. They are quite cheap, on AliExpress you can get 10 for 2 Euro or less (each having 8 outputs) and they can be cascaded by connecting the last output to the next IC's input.

ciapamusche, 2019-12-17:
@mlsomers yes…. the shift register is a good solution, Also a multiplex. 
There are several project which use them for this purpose.

In my opinion it is a very good enhancement… because in live situation you have an immediate visual feedback. Also if the led are insensitive to GUI modifies… doesnt matter.

@alf45tar , irig stomp (the footswitch of Amplitude) react when you modify the pedals in the graphic interface. So i think there is a 2 way data interchange.. but i dont know if its a midi protocol….

alf45tar, 2019-12-17:
@ciapamusche do you mean Irig Stomp or Irig Stomp I/O?
The original Irig Stomp is connected to via an audio jack and there is not MIDI involved.

ciapamusche, 2019-12-17:
yes. Stomp I/O.

alf45tar, 2019-12-22:
Implemented 6 leds status. It works only for momentary 1 and latch 1 for the moment.

closed 2019-12-31 by alf45tar