alf45tar/PedalinoMini #50: when I update found lot of error

jh800624, 2019-12-18:
Sorry I am a code Newbie.
when I update found lot of error 
----------

DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AppleMIDI> 1.6.1
|-- <Battery Sense> 1.1.2
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <Bounce2> 2.52
|-- <AsyncTCP> 1.1.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ESP8266_SSD1306> 4.1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <MD_UISwitch> 2.1.0
|-- <MIDI Library> 4.3.1 #4daf363
|-- <OSC>
|-- <RemoteDebug> 3.0.5
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ResponsiveAnalogRead> 1.2.1
|-- <BLE-MIDI> 0.0.1 #01edf0c
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <Preferences> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <Ticker> 1.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
Building in release mode
Linking .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Building .pio/build/esp32doit-devkit-v1-wifi/firmware.bin
Retrieving maximum program size .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Checking size .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [==        ]  17.0% (used 55560 bytes from 327680 bytes)
PROGRAM: [====      ]  43.1% (used 1356729 bytes from 3145728 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM4
Uploading .pio/build/esp32doit-devkit-v1-wifi/firmware.bin
esptool.py v2.6
Traceback (most recent call last):
  File "/Users/jh/.platformio/packages/tool-esptoolpy@1.20600.0/esptool.py", line 2959, in <module>
[... 891 lines cut ...]
  File "/Users/jh/.platformio/packages/tool-esptoolpy@1.20600.0/esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/Users/jh/.platformio/packages/tool-esptoolpy@1.20600.0/esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/jh/.platformio/penv/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/jh/.platformio/penv/lib/python2.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port COM5: [Errno 2] No such file or directory: 'COM5'
*** [upload] Error 1
===================================================== [FAILED] Took 10.73 seconds =====================================================

Environment               Status    Duration
------------------------  --------  ------------
esp32dev                  FAILED    00:00:23.916
esp32doit-devkit-v1       FAILED    00:00:25.315
esp32doit-devkit-v1-ble   FAILED    00:00:10.955
esp32doit-devkit-v1-wifi  FAILED    00:00:05.292
ttgo-t-eight              FAILED    00:00:23.830
ttgo-t-eight-ble          FAILED    00:00:10.797
ttgo-t-eight-wifi         FAILED    00:00:10.728
================================================ 7 failed, 0 succeeded in 00:01:50.833 ================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

jwyse, 2019-12-18:
> raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
> serial.serialutil.SerialException: [Errno 2] **could not open port COM5**: [Errno 2] No such file or directory: 'COM5'
> *** [upload] Error 1

Edit platformio.ini to set the correct path/COM port (ex: COM3) for your connected microcontroller. 

You don't need to build/flash all configurations, just the one matching your device and desired config (ex: esp32doit-devkit-v1-wifi).

jh800624, 2019-12-18:
How do I found my ESP32 path/COM ?
I usd OSX 10.14.6 very grateful

alf45tar, 2019-12-18:
Use

> platformio device list

or

> pio device list

https://docs.platformio.org/en/latest/userguide/cmd_device.html

davisaminaden, 2019-12-18:
How to flash only my board and ignore others?
My board is ttgo t-eight with built in display, bluetooth and wifi

alf45tar, 2019-12-19:
platformio run -e ttgo-t-eight -t upload

closed 2019-12-31 by alf45tar