alf45tar/PedalinoMini #51: pinout for switch profile

romakrau, 2019-12-23:
Hi, thank you for the led extension. Unfortunatly we lost the pinout for the right and left button, also for the display mode button. Can you give me a hint to have this button extern (gpio 23 e.g.). Greetings Roman

alf45tar, 2019-12-24:
Display mode button is the built-in BOOT button (GPIO_NUM_0) and it is not changed.
There is not too much GPIOs available: 23, 1 and 3.

You can try to play a little bit with the RIGHT_PIN and LEFT_PIN definition into Pedalino.h

>#define FACTORY_DEFAULT_PIN   GPIO_NUM_0
>#define RIGHT_PIN             GPIO_NUM_23
>#define CENTER_PIN            GPIO_NUM_0
>#define LEFT_PIN              GPIO_NUM_23

Cheers
alf45tar

romakrau, 2019-12-24:
Thank you for your quick responce. I can't see GPIO 1 and 3, I can see only that GPIO 2 is not in use. Is this correct or will I damage the board.
Greetings
Roman

alf45tar, 2019-12-24:
Yes, you can but consider that is connected to the board led.

closed 2019-12-31 by alf45tar