alf45tar/PedalinoMini #52: Midi data is not shown on the oled display!

ukudos, 2019-12-24:
Hello, first I want to thank you for your great project. I came across your project while searching the internet for a wireless midi controller for my guitar setup.
I have to say that I have no programming experience. I have now read so far that I have (thanks to your information) made the PedalinoMini work so far. However, I encounter difficulties that I cannot manage alone. Maybe you can help me ...?
I use the Espressif ESP32 DevKit C V4.
In PlatformIO I selected the model: env: esp32dev under ProjectTasks, ran build, upload and UploadFileSystemImage.
Wifi or BLE connections work just fine. However, the OLED (SSH1106 set in platform.ini) shows no midi values ​​when I 
use buttons / switches or potentiometers, only the letter.
But the terminal shows me all relevant data (channel, note on/off, CC, velocity ...).
Any idea what the mistake could be
Thank you again for this unique project and I wish you a nice Christmas!

alf45tar, 2019-12-25:
Each pedal show its name when pressed. By default pedal name is A, B, C, D, E, and F.
If you want to show MIDI event just empty the pedal name via web interface and the relevant data will be shown.

PedalinoMini is also able to show MIDI events received by any interface.

ukudos, 2019-12-25:
Thanks a lot, works perfectly!

closed 2019-12-25 by ukudos