alf45tar/PedalinoMini #53: Trouble getting osc to  work

dbosnyak, 2019-12-30:
I cant seem to get any osc messages to be received.  I added debug code on line 423 of UDPMidiin.h

void oscOnPacket(AsyncUDPPacket packet) {
  DPRINTLN("UDP Packet received");

and it never gets triggered when i send an osc message to the port 8000 of the ip address of the esp32.
I created a whole new program with just the one async udp receiver and it works ok, not sure if it is some interaction with the second async udp receiver for the rtpmidi?  Anyway, have you had it successfully receive osc?  My own program is super simple but it does receive, so I know my osc transmitter is working.  I activated all the osc switches in the interfaces section of the webui.

alf45tar, 2019-12-30:
I added a bug when moved from WiFiUdp to AsyncUDP.
I just release the fix.

Thanks
al45tar

dbosnyak, 2019-12-30:
Can you please tell me what commit fixes the bug?   I am compiling with blynk off.

alf45tar, 2019-12-30:
Every commit dated December 30th should fix OSC message receiving.
May I ask which is your application for PedalinoMini?

Thanks
alf45tar

dbosnyak, 2019-12-31:
Thanks I have it working now, or at least it seems to receive my osc messages.  Great project!  I am trying to use it to get midi and other signals (via osc) into and out of my diy eurorack.  I would like to convert at least one of the pedal inputs into a trigger output, so I can send midi clock received from the rtp port into the eurorack, which will allow me to sync the analog with the daw.  I am also going to expand the osc part a bit so some of the stuff you are doing through blynk could be done through the osc port as well - then you could use any of a huge number of osc controllers with it (like lemur or touchosc on ios/android, or use some of the osc senders/receivers in maxforlive).

dbosnyak, 2019-12-31:
Is interfaces[PED_DINMIDI].midiOut the same as 'Legacy MIDI' in the web interface?

closed 2019-12-31 by dbosnyak

alf45tar, 2019-12-31:
Yes, it is.

wrpooley, 2020-03-06:
I would love to know more about getting OSC to function.  So far no luck with any RTP-Midi or OSC.  Is there something to the routing I should know?  Does PedalinoMini just mirror whatever midi command it's generating to all interfaces?  I'm familiar enough with the OSC protocol to configure it in Reaper so I don't think it's pilot error.