alf45tar/PedalinoMini #54: using momentary switch as toggle switch?

ukudos, 2019-12-30:
Hi alf45tar,
is there the possibility to use momentary switches as toggle switches?
The background is: I would like to switch effects on or off in BiasFX .
Unfortunately, I can't manage it, because momentary switches send messages when pressed and when released (e.g. Note On/Off or 127 or 0 ...).
It would be great if you press once: effect on and the second time you press: effect off.
Maybe I'm making a mistake, but I haven't figured it out yet ...

Best regards
ukudos

alf45tar, 2019-12-30:
BiasFX reacts only on Control Change. For TOGGLE effects the CC value doesn't matter.
PedalinoMini sends MIDI Value 1 on press and MIDI Value 2 on release.

I let you know how to proceed asap.

alf45tar, 2019-12-30:
@ukudos update PedalinoMini to the latest version just released

ukudos, 2019-12-31:
Hello alf45tar,
Thanks for your quick reply!
It works great !!!

Just an idea: wouldn't it be cool if you could use a momentary switch to e.g. Control parameters !? As examples, I think of T.C. Electronics ("MASH" switch) or Digitech ("Ricochet")?

Thank you for your attention and your great project!
ukudos

bauerbyter, 2020-01-06:
Hey, first of all, your project is freaking awesome (wanted to build myself something like this, to see this is already available). But I have the same question as ukudos and maybe I dont get it. 
What would be cool:
Press Momentary Switch => CC  xx 127-> activate delay (for example)
Press Momentary Switch (the same) =>  CC xx 0 -> deactive delay.
Is this alreay implemented and I cannot find it? 

Thanks in advance, Bauerbyter

alf45tar, 2020-01-06:
Yes, there is somenthing similar.
If you enable Double Press and Long Press too Pedalino can send Midi Value 1 on Single Press, Midi Value 2 on Double Press and Midi Value 3 on Long Press.

One momentary switch can send 3 different values.

Which is the device that requires 127 for activate and 0 to deactivate?

bauerbyter, 2020-01-06:
Yes I found that features, but thats not exactly what I mean.
I am guitarplayer and as far as I know, every software, amp or rackmodule is using this.
0 = off 
everything above 64 = on
For example: TC G-Major & Boss Katana (thats what I own) but as far I know, these is commonly used (maybe I am wrong) also a lot Footcontrollers you can buy, do this => send CC 127 and set LED on. Second press => send CC 0 and set LED off.

bauerbyter, 2020-01-06:
marosell is also doing this in this video, but he needs two buttons/switches for it.
https://www.youtube.com/watch?v=BEKNbTDvX7c&feature=youtu.be (around 1:05)

alf45tar, 2020-01-06:
Update to the latest version just released.
First press send MIDI Value 1 and second press send MIDI Value 2.

bauerbyter, 2020-01-06:
Wow this was fast and it is working! just tested it. Thank you very much!

closed 2020-02-10 by alf45tar