alf45tar/PedalinoMini #56: Booting modes

Hans68, 2020-01-05:
Hi,
I am back after some weeks busy in other projects. I am happy to see that pedalinomini development is going on, thanks a lot.
You mentioned in your Readme file:
"The last booting mode (1-5) is selected if you don't press any button on boot."
I have loaded esp32 dev kit and selected by pressing En and boot button either ble or Wi-Fi mode. 
This worked, but after switching power off and on , it starts always again with normal mode. 
Beside that, it would be helpful to explain with some words the specific functionality of each mode, e.g. what the difference is between ble mode and ble access mode.
Thanks a lot and Cheers
Hans

alf45tar, 2020-01-06:
Hi Hans,
eeprom is changed over time. I suggest a reset to factory default.
Check the console for error messages or add here and I will have a look.

I know PedalinoMini is not so well documented. Here in the following more details. 

Thanks
alf45tar

1. Normal
BLE and WiFi are enabled.
On boot PedalinoMini will start the WiFi procedure (connect to last AP -> SmartConfig -> WPS -> Access Point).
After boot PedalinoMini will wait for BLE-MIDI connection.

2. Bluetooth Only
WiFi is disabled.
PedalinoMini is waiting for BLE-MIDI connection.

Mode 3 "WiFi Only"
On boot PedalinoMini will start the WiFi procedure (connect to last AP -> SmartConfig -> WPS -> Access Point).

Mode 4 "Access Point with Bluetooth"
PedalinoMini will skip the WiFi procedure and create a WiFi access point.
BLE-MIDI is enabled.

Mode 5 "Access Point without Bluetooth" 
PedalinoMini will skip the WiFi procedure and create a WiFi access point.
BLE-MIDI is disabled

Hans68, 2020-01-06:
Thanks for the quick response as always !
I only get stable operation, if I either build and load env:esp32doit-devkit-v1-ble or env:esp32doit-devkit-v1-wifi.
Loading env:esp32doit-devkit-v1 I have following problems:
1. In wifi mode or normal mode I get a reboot as soon as I try to give any command from web-interface to configure banks/ pedals etc. (see attached serial monitor excerpt). 
Sometimes, this reboot happens without any action.
2. In ble mode connecting is possible, but the blue LED is only very dark flickering, while after loading pure ble mode the LED is clearly and bright blinking.
I did reset to factory mode, this lesson I learnt already some months ago ;)
Cheers
Hans

[pedalino error reboot.pdf](https://github.com/alf45tar/PedalinoMini/files/4026942/pedalino.error.reboot.pdf)

alf45tar, 2020-01-06:
esp32doit-devkit-v1-ble compile only ble. WiFi will not be included into the firmware.

esp32doit-devkit-v1-wifi compile only wifi. BLE will not be included into the firmware.

The best way is to use esp32doit-devkit-v1 and select WiFi **or** BLE using the boot mode.
Using WiFi **and** BLE toghether is currently unstable because there is not enough memory.

Hans68, 2020-01-07:
Thanks, I would prefer your recommended "best way", but I have following effects as partly already stated by using esp32doit-devkit-v1:
1. After power on, Pedalino starts always in "normal mode", independent what I have used before power off.
2. If I select "BLE only" by using EN/Boot procedure, I can connect by midimittr, but ESP shows strange behaviour (flickering blue LED) (see attached file)
3. After power on, Pedalino starts in normal mode and is in most cases very fast synchronizing to my wifi network. If I then change to "wifi only" mode by using EN/Boot procedure, it does not synchronize to my network and goes to default mode with smart config etc. (see two attached files).

Your comment is appreciated.
Cheers
Hans
[normal boot and change to BLE mode.pdf](https://github.com/alf45tar/PedalinoMini/files/4030182/normal.boot.and.change.to.BLE.mode.pdf)
[Start with reboot in normal mode.pdf](https://github.com/alf45tar/PedalinoMini/files/4030183/Start.with.reboot.in.normal.mode.pdf)
[Start without reboot in norml mode.pdf](https://github.com/alf45tar/PedalinoMini/files/4030184/Start.without.reboot.in.norml.mode.pdf)

alf45tar, 2020-01-07:
Flickering of blue led when a BLE client is connected is by design. If you disconnect the client the blue led will stop to flicker.

Thanks for the logs. I will check in deep them but the first think that I noted is that you are not using the latest firmware.

Hans68, 2020-01-07:
Thanks !I am quite sure that I am using the latest firmware, but most probably you can see it in the logs.Concerning flickering of LED: I was surprised that the LED was bright and rhythmic blinking when I used env: esp32....-ble and it was only dark and not rhythmic flickering when I selected 'ble only' mode by En/boot procedure.If this is not a bug , no problem for me.Cheers HansVon Samsung-Tablet gesendet

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/alf45tar/PedalinoMini/issues/56?email_source=notifications\u0026email_token=ALR3NU7OKFITVW7DT6KZSMTQ4RZJ5A5CNFSM4KC4XKX2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIIXFVA#issuecomment-571568852",
"url": "https://github.com/alf45tar/PedalinoMini/issues/56?email_source=notifications\u0026email_token=ALR3NU7OKFITVW7DT6KZSMTQ4RZJ5A5CNFSM4KC4XKX2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIIXFVA#issuecomment-571568852",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

alf45tar, 2020-01-07:
The latest version just released fixed the blue led.

Hans68, 2020-01-08:
I got an ERROR message during bulid (see attachment)
Cheers
Hans

[Build failed with newest version.pdf](https://github.com/alf45tar/PedalinoMini/files/4035445/Build.failed.with.newest.version.pdf)

alf45tar, 2020-01-08:
Update project libraries or just delete the case in the source file

Hans68, 2020-01-09:
Thanks, great hint !
 I updated Poject libraries (ESP) and also my other problems I mentioned last days are solved. 
There is one  new topic:
I cannot use "Access Point with Bluetooth" because  I get permanent reboots. All other booting modes are working fine. (see attached log)
Cheers 
Hans

[Permanent reboot in AP with BLE.pdf](https://github.com/alf45tar/PedalinoMini/files/4039925/Permanent.reboot.in.AP.with.BLE.pdf)

alf45tar, 2020-01-09:
"Access Point with Bluetooth" mode required WiFi and BLE but the memory is not enough.
If you check the log you have around 36K of free heap (Internal Free Heap 36888). For reliable operations it is requested 80-100k.

Hans68, 2020-01-09:
That's a pitty, because this would be my preferred mode. Can I delete something else to get more free heap? I do not need BLYNK for example. 
Thanks and Cheers
Hans

alf45tar, 2020-01-09:
Yes, it is a pitty.

A temporary workaround could be:
- configure PedalinoMini via web interface
- recompile and upload firmware with - D NOWEBCONFIG and -D NOBLYNK

You will have 60K free that it is enough without the web server for the web interface.

Remember to download the latest firmware version.

Hans68, 2020-01-10:
Thanks,  will test tomorrow and let you know.Cheers Hans Von Samsung-Tablet gesendet

Hans68, 2020-01-11:
Hi,
Thanks a lot, but your proposed solution does not solve my problem.
My preferred solution would be to use the Access Point with BLE to switch fast between Webconfig and BLE operation. With other words, I want to enter different Pedalino Configs and directly test it with my BLE device, in my case JamUp via BLE-MIDI or midimittr.
I do not need BLYNK, Standard Midi, WiFi Midi and access to existing WiFi Network, Access Point 192.168.4.1 is preferred due to independance of location.
If there is no other option, I have to switch in "Booting modes" between WiFi only and BLE only.
Cheers
Hans

closed 2020-01-22 by Hans68