alf45tar/PedalinoMini #57: Extra Buttons

bauerbyter, 2020-01-06:
Hello,
again: your Project is absolute awesome, thanks for all that work.
I am at the moment digging through your code  and found the PEDALS define in the config.h and in the readme you are writing "Each port can connect 1 expression pedal or up to 5 foot switches for a maximum of 30 foot switches"
So I am building myself a footcontroler but I wanted to have 15 Switches (and one espression Pedal) but I dont really get how to add more physical switches to your hardware. 

Thanks and greetings!
Bauerbyter

alf45tar, 2020-01-06:
Use a voltage ladder as indicated into the schematic.

bauerbyter, 2020-01-06:
A got it, sorry was blind! thanks for that.

closed 2020-01-06 by bauerbyter

alf45tar, 2020-01-06:
Please share your project when done.

bauerbyter, 2020-01-06:
Shure, will do!