alf45tar/PedalinoMini #58: [Question] setup ipMidi between two PedalinoMini

bauerbyter, 2020-01-07:
Hey,
what has to be done to setup two PedalinoMinis to communicate over ipMidi. 
Same network and activate IN and OUT for ipMidi, is clear. Something else?

Thanks again, cannot thank you enough.

alf45tar, 2020-01-07:
You are right.

Same network because ipMidi is a multicast protocol.
Enable ipMidi OUT on sender and ipMidi IN on receiver.
Enable Legacy MIDI OUT interface on receiver to route to a standard MIDI equipment.

bauerbyter, 2020-01-07:
Thanks, I will try this out the next week and I also will try if its working bi-directional. 
The Big picture (if you are interested) is something like this:
Floorboard sends PC, RX-Box is getting this and send with legacy midi the Amp and the Multieffect to specific channel/preset .
The Multieffect will send back which effects are active in this "preset" -> RX-Box sending -> floorboard is getting this information and setting the specific CC-Switch to active or passiv. 
So imagin you have 4 Channels (= 4 PC Switches), Bank Up/Down Buttons and 4 CC Switches (in total 10 switches for the small version ^^)
So for example you click the first PC-switch (=> sending PC 0) this is the clean channel and you have delay and reverb active on it, so the LEDs unter these matching CC-Switches are lighting up and when you press on of these, the accordingly effect is turned of ( => CC xx 0) when you click one of the other CC-Switches they will turn a effect on (CC xx 127, Chorus for example). 

I am at the moment going through your code and looking for how I can contribute and help with that, so what would be great would be the possibility to add Switch multiplexer and LED multiplexer (and can control the LEDs in the web).

ukudos, 2020-01-15:
Hey bauerbyter,
did you get ahead with the LED feedback?
That would be a fantastic extension for the pedalino project ... in my opinion.

bauerbyter, 2020-01-16:
working on it. But dont have plenty of time the next days.

alf45tar, 2020-01-25:
Do you know the sequence feature? You can store sequence of MIDI messages inside Pedalino and send   on footswitch press. A sequence can activate all the requested effects on a multieffect.

alf45tar, 2020-01-25:
In the latest version there is also a new feauture.
When PedalinoMIni receives a NoteOn/CC/PC corresponding to a pedal it turn on (or off) the corresponding led.

bauerbyter, 2020-01-25:
nice! this is awesome

ukudos, 2020-01-28:
@alf45tar
is the new feature also available via bluetooth, ipad and BiasFX? Unfortunately, it didn't work during my attempts.
I have two more questions for you that are somewhat off topic.
1. Is it possible to rotate the OLED ssh 1106 display by 180 degrees by software? During my search on the net I found some approaches, which unfortunately I cannot implement due to my lack of knowledge. Maybe you have a tip on how I can implement this ...?
2. I want to operate the PedalinoMini with 4x 1.5 volt batteries (ESP32 DOIT devkit).
In the build flags I set "Battery" and via a voltage divider, which delivers 1.1 volt output to pin 23. Unfortunately without function!
Do you see a way to implement a voltage display for the ESP32 DOIT devkit?

greetings
ukudos

alf45tar, 2020-01-29:
The new feature should works on any interface (USB, DIN, BLE, ipMIDI, rtpMidi) .
Remember to enable the IN switch on the interface tab.

To rotate the OLED just comment the following lines
https://github.com/alf45tar/PedalinoMini/blob/90b3d1a80dbac6f018e5117893b94b59def1a59b/src/DisplayOLED.h#L974
https://github.com/alf45tar/PedalinoMini/blob/90b3d1a80dbac6f018e5117893b94b59def1a59b/src/DisplayOLED.h#L1027

Never try the battery on ESP32 DOIT but I don't see any specific issue. Are you using two resistors of the same value? There is a ratio of 2 in the code.

https://github.com/alf45tar/PedalinoMini/blob/90b3d1a80dbac6f018e5117893b94b59def1a59b/src/DisplayOLED.h#L1030

The code is expecting a voltage between 3V and 4.2V like in any 3.7V lithium battery.

https://github.com/alf45tar/PedalinoMini/blob/90b3d1a80dbac6f018e5117893b94b59def1a59b/src/DisplayOLED.h#L36

ukudos, 2020-01-30:
Many thanks for your response!
I still have to deal with the voltage display ...
The new function (LED feedback) does not yet work with the combi: iPad, BLE or USB-Midi. If a CC event is executed in BiasFX via USB, the ProMicro also receives data. Unfortunately I have not yet found out which data it receives. In e.g. LogicX it runs smoothly and it's a really great extension.

Thanks again for your great project, your patience and your willingness to support.

Best regards
ukudos

alf45tar, 2020-01-30:
Are you sure that BiasFX is able to send MIDI messages?
The MIDI interface is called "Active MIDI Input" (see
https://help.positivegrid.com/hc/en-us/articles/360027975151-How-to-Use-MIDI-Control-in-BIAS-FX-2-Standalone-Version-). I suppose no MIDI output.

I will test more in depth all the possible combinations.

Thanks for experimenting with my project
alf45tar

ukudos, 2020-02-10:
Hello alf45tar,
it's me again...
I'm getting a little embarrassed to ask for help.
The voltage display gives me a little headache.
I followed your instructions:
`Battery bat(4000, 6000, BATTERY_PIN);
`
`  bat.begin(3300, 2);     // , &sigmoidal`
(Try with and without sigmoidal)

The result is: I get a value of 6593mV.
No matter whether with battery or via USB and the value remains constant, even with decreasing battery voltage.
Input voltage via battery is 6 volts at VIN (voltage divider with 2x1k resistors and in further experiments with 2x10k!)
Do I have to make changes in the Battery.h file?

Regarding the midi-out capability of BIAS: I asked the Positive Grid support about 2 weeks ago. Unfortunately no response has been received so far. But I also think that you are right: no midi-out support!

Best regards
ukudos

closed 2020-03-11 by alf45tar