alf45tar/PedalinoMini #60: configuration lost by changing to "BLE only"

Hans68, 2020-01-12:
Hi,
If I make a configuration in "WiFi only" mode which is C.04 and change than to "BLE only" I get again A.01. 
Can you give me  a hint what I am doing wrong? Please see log attached.
Cheers
Hans

[change to Profile A and Bank 1 after boot in BLE.pdf](https://github.com/alf45tar/PedalinoMini/files/4050881/change.to.Profile.A.and.Bank.1.after.boot.in.BLE.pdf)

Hans68, 2020-01-13:
I have tested further and uploaded first env: esp...-wifi, made my configuration in WEB Page and loaded afterwards env: ESP....-ble. This shows the same behaviour, that means change to A.01 and in addition I recognized, that even A.01 configuration was reset to factory default.
Your help is appreciated!
Cheers
Hans

alf45tar, 2020-01-16:
It is not a bug. It is a feature and you not doing anything wrong.
I am joking. It is by design but I not sure it is the best solution.
PedalinoMini will start with profile A at every reboot.

The code to save the current profile it is already in the files.
If you add the following line

> saveProfile = true;

somewhere the current profile will be saved immediately.

Let me know when do you suggest/prefer to save current profile for next reboot.
When you press the Save button or every time you change the current profile on top left corner of the web interface?

Hans68, 2020-01-17:
Thank you!I think it is s must to keep the configured profils/banks/pedals saved and not set it back to default during boot. Or is there any other solution to configure the profiles for BLE mode?In my point of view the only way to do this is to select "Wi-Fi only" and configure the profils and afterwards switch to "BLE only" and use it.You mentioned to add the line "saveProfile = true;" somewhere in the code. Could you please be a little bit more precise (in .ini or somewhere else?).Currently I am using an old version of end of October timeframe and with that version profiles remain saved after switching from Wi-Fi to BLE and in addition I could select 3 different profiles be two keys connected to Tx2 and Rx2. Is this feature cancelled or is it still available to be selected instead of the LED's via U11?I do not want to use old versions because I would be decoupled from any bug fixes and improvements you are doing in future.CheersHans Von Samsung-Tablet gesendet

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/alf45tar/PedalinoMini/issues/60?email_source=notifications\u0026email_token=ALR3NU7UQR6QEY6ST2KOX4LQ6DL6ZA5CNFSM4KFY3AVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJFXWXI#issuecomment-575372125",
"url": "https://github.com/alf45tar/PedalinoMini/issues/60?email_source=notifications\u0026email_token=ALR3NU7UQR6QEY6ST2KOX4LQ6DL6ZA5CNFSM4KFY3AVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJFXWXI#issuecomment-575372125",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

alf45tar, 2020-01-17:
With the last versione current profile is saved when Save buttom is used.

Use of GPIO is defined in Pedalino.h. They can be changed within certain limits. 

#define FACTORY_DEFAULT_PIN   GPIO_NUM_0
#define RIGHT_PIN             GPIO_NUM_23
#define CENTER_PIN            GPIO_NUM_0
#define LEFT_PIN              GPIO_NUM_23
#define SERIAL_DATA_PIN       GPIO_NUM_16   // DS
#define CLOCK_PIN             GPIO_NUM_5    // SH_CP
#define LATCH_PIN             GPIO_NUM_17   // ST_CP
#define USB_MIDI_IN_PIN       GPIO_NUM_18
#define USB_MIDI_OUT_PIN      GPIO_NUM_19
#define DIN_MIDI_IN_PIN       GPIO_NUM_15
#define DIN_MIDI_OUT_PIN      GPIO_NUM_4
#define BATTERY_PIN           GPIO_NUM_23

Hans68, 2020-01-18:
Thanks a lot, issue solved and closed.
I opened another one concerning LATCH mode.
Cheers
Hans

closed 2020-01-18 by Hans68