alf45tar/PedalinoMini #61: Jog wheel support is unclear

jypma, 2020-01-13:
The README claims support for jog wheels (rotary encoders), but it's unclear to me how they're supposed to be hooked up (the example schematics don't include a jog wheel). Additionally, I couldn't find much code implementing them either, other than an empty `switch` [here](https://github.com/alf45tar/PedalinoMini/blob/d1b43b7157e47c2c0a58926b92daf96a747bf25d/src/Controller.h#L1368).

Are jog wheels / rotary encoders implemented? It'd be great to do something akin to the BCR2000's functionality, sending different CC values for left/right, potentially affected by speed/accelleration as well.

alf45tar, 2020-01-13:
Jog wheels not yet implemented

jypma, 2020-01-13:
Thanks for clearing that up! Do you have any ideas to share for direction, in case someone (me or otherwise) would be willing to pick it up? I'm thinking important areas could be:
- Suggested electrical connection. Perhaps a resistor ladder like was done for the multi-switches, wich a few capacitors tacked on for de-bouncing? We could have 2 encoders on 1 jack then, using 2 virtual inputs for each.
- Suggested MIDI data to send. I couldn't find an authoritative source on this.

alf45tar, 2020-01-16:
I think the the encoder module KY-040 should be used because it is cheap and easy to find.
To fully connect it we need 2 PedalinoMini ports_ one for the rotary input and one for the push button.
I never tried yet but my first try could be: 

Port 1:
- CLK to A1
- DT to D1
- GND to GND

Port 2:
- "+" to A2
- SW to D2
- GND to GND

Rotary knob and push button could send any supported MIDI message (PC, CC, note, pitch bend, channel pressure) configured in the web interface.

My 2 cents
alf45tar

alf45tar, 2020-01-22:
Added partially support of jog wheel. For the moment the jog wheel can be connected as per schematic. Rotation and switch reading is correct but it is not sending any MIDI message. 

EEPROM is changed. Remember to reset to factory default.

alf45tar, 2020-01-23:
Now jog wheel can be used to sending MIDI messages

closed 2020-01-24 by alf45tar