alf45tar/PedalinoMini #63: Touch sensitive behaviour Pin D18

Hans68, 2020-01-29:
Hi alf45tar,

first of all again many thanks for this great project and your continous support. My Pedalino is nearly finished and working great. Attached a picture of my solution.
Now to the issue: 
I recognized a sensitive behaviour if I touched the non metal housing (glass fibre plastic) close to the region where the Pro micro is located. I got strange displays on the OLED. 
I solved this problem by adding a 100k resistor from D18 to ground. Perhaps helpful also for others.

Cheers
Hans
[image]

alf45tar, 2020-01-29:
Thanks for sharing your built. If you want to add photos from different point of view (inside included) it will be very appreciated.
Did you customized anything (apart the 100k resistor)?

Hans68, 2020-01-30:
Hi,

Please find attached some more pictures and the circuit diagram.

Cheers
Hans

[Hans68 pedalinomini.pdf](https://github.com/alf45tar/PedalinoMini/files/4135089/Hans68.pedalinomini.pdf)
[image]
[image]
[image]
[image]
[image]
[image]

mknerr, 2020-01-30:
@Hans68 Nice work. Do you know yet what kind of battery life you are getting? And if you do, what mode are you using, for reference. Thanks!

Hans68, 2020-02-01:
Hi,
I use "WiFi only" for configuration and "BLE only" mode for operation.  In
BLE mode with all LED's on I got a current of app. 280 mA measured directly
at the 18650 Lion battery with
2.600 mAh. So an operation of  >8 hours should be more than enough for one
Gig.
Cheers
Hans

mknerr, 2020-02-01:
Great. Thanks for the info.

closed 2020-02-03 by Hans68