alf45tar/PedalinoMini #64: Hold Boot button does not change from live view

jimhiggs, 2020-01-31:
I am just starting to build and have uploaded to my esp32 and connected 10k tieups to A1-A6 and connected the display. I am booting into wifi and works well. The oled displays A.01 and  pressing boot button changes to B1. Pressing boot twice just seems to go from A.01 to B.01 and then C.01 (not sure if that is how it does the 'back one'). Holding boot just keeps cycling through A, B, C. I can't find any way to get to the home screens. The only time I see anything different (aside from the boot screens) is when I set masterclock on the web app. Setting master clock and start also resets my esp32 regularly.
How can I switch to other screens? 
Also when I operate a switch on say A1 I don't see anything should I?
I have an alixpress esp32 so it could be iffy.

Apologies if these are silly questions, I have been searching for answers for several hours. I am also very grateful to the devs, i couldn't believe I would find an open source solution and one of such high quality too.

jimhiggs, 2020-01-31:
I tried another ESP32, but from the same source. Same problem. But I was able to get into the home screens by sort of sliding my finger off the boot button so a very quick on off? Normal use of the boot button seems to be picked up as several presses. When It first boote after upload the boot button stepped nicely through abc but after a reboot it returned to being problematic. A question: is it possible to re-assign the boot button to another pin eg D2? so that the button can be externally mounted?

alf45tar, 2020-01-31:
Single press of BOOT button to move to the next profile. If you press when the current profile is C it moves to profile A.

Double press of BOOT button to move to the previous profile. If you double press when the current profile is A it moves to profile C.

Long press of BOOT button to switch between live performance display and scrolling pages where configuration parameters (the device name, the IP address etc.) are displayed.
Long press definition is longer than 500 ms but not over 1 sec. You change them in Options.

If you keep pressed for more than 1 sec the profile change every 500 ms to the next one.

You can assign the feauture of BOOT button changing this line of code
https://github.com/alf45tar/PedalinoMini/blob/90b3d1a80dbac6f018e5117893b94b59def1a59b/src/Pedalino.h#L52

jimhiggs, 2020-01-31:
Thanks for the response. I understood what the actions of the Boot button were from reading the instructions on the github page, but my button acts strangely. I will try re-assigning the CENTER_PIN. Thank you.
Also on the first esp32 I tried I didn't realise I had to select one of the versions so I think it basically installed each in turn, so I assume I would be left with the last and wrong one for the ttgo board. 
On the 2nd ESP32 I tried I tried just esp32doit-devkit-v1-wifi, then esp32dev, no change in boot key behaviour but the web display is not formatted, like the css is missing?? I did both an upload and an upload_file_system_image.

jimhiggs, 2020-01-31:
Forget my questions, I will get some more esp32's on order. If nobody else is having similar problems i guess it is probably the boards.

jimhiggs, 2020-01-31:
Just found the reason. When  I ran the web app and went to options all the times where set at 0. I changed them to the same default values as in pedalino.h and saved and the button now works as expected. The web app looks ok again so not sure what was going on there.

jimhiggs, 2020-01-31:
I would like to ask though why the boot pin is used to change bank and page rather than a pin that can be fitted through the case? I tried D2 but it is connected to the LED I believe. I do see you have rather run out of pins. Can you briefly explain what the right and left pins are for would GPIO_num_23 be available for the boot pin function?

jimhiggs, 2020-01-31:
Is there a forum somewhere or FB group rather than asking for advice in the github issues Q.

alf45tar, 2020-01-31:
github is the only way to obtain support.
LEFT_PIN and RIGHT_PIN are used for TTGO board. GPIO_NUM_23 is the only pin not yet used.
You can use it to replace BOOT but I will not guarantee I will not use for other feauture later.

jimhiggs, 2020-01-31:
OK, Thanks for the help. It is a fantastic design, I am planning on building 2. One using 6 TRS jack inputs as a general analog to midi converter and one with 6 or so momentary switches as an extend er for my DX stomp pedal. That is the reason I started looking but this is so much more. Thanks again.

alf45tar, 2020-01-31:
Sharing project photo will be appreciated.

closed 2020-01-31 by alf45tar