alf45tar/PedalinoMini #66: TC Helicon Ladder Switch

mknerr, 2020-02-01:
TC Helicon makes a very robust 6 button ladder switch, however, it uses all 10k resistors. Could pedalino be made to utilize this foot switch?

alf45tar, 2020-02-01:
I added support for TC-Helicon Switch 6 but I cannot test it because I don't own it. I created an equivalent circuit I found on internet for my testing. Do you have one to test for me? 

Select "Ladder 6" as Pedal Mode and connect TC-Helicon inverting Tip and Ring.
**T**ip of TC-Helicon must be connected to Ring on PedalinoMini.
**R**ing of TC-Helicon must be connected to Tip on PedalinoMini.
**S**leeve of TC-Helicon must be connected to Sleeve on PedalinoMini.

mknerr, 2020-02-02:
Amazing! Yes I have one I can test.

mknerr, 2020-02-10:
I've had a couple of tries at this, and it just does not seem to be working properly with my switch6. 
I am using a standard cable, but I have switched tip and ring on my pcb.

I do not get a response out of switches 1-4. Switch 5 fires bank 6, and Switch 6 fires bank 1.

Not sure what to do at this point. I did not erase the ESP before updating, if that matters.

Thanks again for including the switch. I think it will be a great option for people.

jimhiggs, 2020-02-10:
I can confirm the ladder 6 is not working for me either. I got hold of a helicon switch 6 yesterday. Mine is a little more erratic. I get nothing from buttons 1 - 4. if i press 4 (nothing) and then 5 i get E5 and if I press 6 and then 5 I get E6 But not every time most times nothing (I named the banks E1 - E10).

alf45tar, 2020-02-11:
I will try a different approach

alf45tar, 2020-02-11:
I added a calibration function for ladder footswitch. PedalinoMini will support virtually any ladder with any resistor values.

@mknerr or @jimhiggs Can you copy and paste the output log of your TC-Helicon Switch 6? Attach it to Pedalino and follow the OLED display instructions.

The log should looks like the following
Ladder 1 Threshold 403 Tolerance 7
Ladder 2 Threshold 419 Tolerance 5
Ladder 3 Threshold 431 Tolerance 5
Ladder 4 Threshold 443 Tolerance 5
Ladder 5 Threshold 456 Tolerance 3
Ladder 6 Threshold 465 Tolerance 3

Thanks
alf45tar

jimhiggs, 2020-02-11:
Gladly but I do not know how to enable or view the log and after I attach the pedal what do I do  to get oled instructions?

jimhiggs, 2020-02-11:
I see after flashing the latest it starts to calibrate fs3 (strange mine is on A5/D5). So I powered down until I know how to turn on/display logging.

mknerr, 2020-02-12:
Here is the output from my test, but I did it a few time, and it wasn't very consistent. The timing of my presses changed the output too, and I wasn't exactly sure when and how long the buttons should be pressed. Should it be only within the time the bar is advancing, or should I hold until the bar starts over for the next switch? Below represents me holding until the bar starts over, because when I did it the other way, the values were all the same.

Ladder 1 Threshold 668 Tolerance 253
Ladder 2 Threshold 663 Tolerance 253
Ladder 3 Threshold 660 Tolerance 242
Ladder 4 Threshold 634 Tolerance 227
Ladder 5 Threshold 577 Tolerance 187
Ladder 6 Threshold 440 Tolerance 187

@jimhiggs To view the log output, your pedalino needs to be connected to your computer running PlatformIO, then select the "Monitor" function in the project tasks section of the pane on the left.

mknerr, 2020-02-12:
Here is another test. Also, the switches are all firing bank 3, except for SW1, which does nothing

Ladder 1 Threshold 652 Tolerance 5
Ladder 2 Threshold 665 Tolerance 1
Ladder 3 Threshold 669 Tolerance 246
Ladder 4 Threshold 650 Tolerance 232
Ladder 5 Threshold 604 Tolerance 187
Ladder 6 Threshold 468 Tolerance 187

jimhiggs, 2020-02-12:
@mknerr
If you could tell me how to log I could test as well

alf45tar, 2020-02-12:
@mknerr Thank you for the testing. Holding until the bar starts over is correct. 
Results are very interesting.
Your threshold values are increasing from 1 to 3 and decreasing from 3 to 6.
Are we sure that TC-Helicon Switch 6 is a resistors ladder?
Also the tolerance between switches is not so big to have a reliable detection.
There is something that I don't know.

jimhiggs, 2020-02-12:
I will take it apart and check but I have measured the resistances between tip/ring, tip/sleeve and ring/sleeve and they match your 6 ladder schematic.

jimhiggs, 2020-02-12:
Button Pressed           |     0     |   1   |   2   |   3   |   4   |   5   |   6
tip/sleeve                   | Mohm | 60k | 50k | 40k | 30k | 20K | 10K
tip/ring                      | Mohm | 10K | 20K | 30k | 40k | 50k | 60k
Ring/Sleeve               |   70k    | 70k | 70k | 70k | 70k | 70k | 70k

alf45tar, 2020-02-12:
It is definitively a resistor ladder as per schematic.
A constant resistor produces a logarithmic response. Logarithmic is very flat and output value are very close.

jimhiggs, 2020-02-12:
Also on my Fluke meter all readings are within I would say 2%

jimhiggs, 2020-02-17:
I just uploaded to the latest. Got all sorts of odd behaviour BUT just rebooted and ran through the switches and got:
Ladder 0 Threshold 945 Tolerance 35
Ladder 1 Threshold 876 Tolerance 31
Ladder 2 Threshold 816 Tolerance 31
Ladder 3 Threshold 752 Tolerance 33
Ladder 4 Threshold 660 Tolerance 47
Ladder 5 Threshold 497 Tolerance 82
And having named banks 1-6 on pedal 5 E1-E6 I am indeed getting the correct pedal,. Will test further.

jimhiggs, 2020-02-18:
Sad part is I can't repeat it. lol. Will keep trying.

jimhiggs, 2020-02-18:
It seems I have to be a lucky or perhaps  its timing in the initial setup of switches as to whether it works or not. Sometimes all pedals work but give me the E1 output. other times just switch 6 works and other times it seems to be correct but gives me a pedal off event as well, eg press 1 get E1, E4. Press 3 get E3 E4. The second note is the same for all buttons but has been different for different resets.  Other times, like now, it is working as it should. Anything I can try in order to narrow it down?

jimhiggs, 2020-02-18:
Here is a log where it failed the first try and worked the second. The reboot is occurring when I reload the /pedals page in the web ap. I am using default_envs = esp32doit-devkit-v1-wifi
[log.txt](https://github.com/alf45tar/PedalinoMini/files/4216944/log.txt)

jimhiggs, 2020-02-18:
I changed the CC's so could see it works; Extract from log:
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   1......Value   1.....Channel  1
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   2......Value   2.....Channel  1
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   3......Value   3.....Channel  1
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   4......Value   4.....Channel  1
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   5......Value   5.....Channel  1
Pedal  5   SINGLE PRESS 
CONTROL CHANGE.....Code   6......Value   6.....Channel  1

jimhiggs, 2020-02-18:
Ok Firstly, I am not a C++ programmer but I have other programming experience. It seemed to me that it was the setting of the kt[i] values that was screwing me up so Just to test, I  removed all the code under if (pedals[p].mode == PED_LADDER6) { and added the values from my successful boot eg:
    for (byte i = 0; i < LADDER_STEPS; i++) { 
       kt[i].value = i;
     }     
     kt[0].adcThreshold = 945;
     kt[1].adcThreshold = 876;
     kt[2].adcThreshold = 816;
     kt[3].adcThreshold = 752;
     kt[4].adcThreshold = 660;
     kt[5].adcThreshold = 497;
     kt[0].adcTolerance = 35;
     kt[1].adcTolerance = 31;
     kt[2].adcTolerance = 31;
     kt[3].adcTolerance = 33;
     kt[4].adcTolerance = 47;
     kt[5].adcTolerance = 82;

I can confirm that with these values, my Pedal6 works as I would expect it to.

alf45tar, 2020-02-18:
@jimhiggs did you make any changes to the hardware connection?

jimhiggs, 2020-02-18:
No, I just have one socket dedicated to the ladder 6 with tip and ring reversed and use a standard trs cable.

alf45tar, 2020-02-18:
Thanks you @jimhiggs. You confirmed that it can works.
I found the error and I will release the fix asap.

alf45tar, 2020-02-18:
New code just released.
Select "Ladder" as Pedal Mode and enable Autosensing. "Ladder 5" and "Ladder 6" has no more sense because PedalinoMini can now support any type of ladder up to 6 switches.
Reboot PedalinoMini and do the ladder calibration (press each switch in sequence until the bar reach zero).
For now the calibration values are not saved into EEPROM. Then the calibratio is requested at each reboot.

jimhiggs, 2020-02-18:
Afraid it does not work for me. Calib values are:
Ladder 0 Threshold 640 Tolerance 5
Ladder 1 Threshold 653 Tolerance 4
Ladder 2 Threshold 646 Tolerance 4
Ladder 3 Threshold 625 Tolerance 11
Ladder 4 Threshold 566 Tolerance 30
Ladder 5 Threshold 430 Tolerance 69

alf45tar, 2020-02-19:
Does it works again with your code?

I had the same issue on GPIO 33 Pedal 6). I changed the GPIO to 32 (Pedal 5) and it worked. I supposed that my GPIO 33 was fried but  I switch back to GPIO 33 (Pedal 6) and it worked too.

I don't know what happened.

jimhiggs, 2020-02-19:
I added the two lines of code after
    if (pedals[p].mode == PED_LADDER && pedals[p].autoSensing) {
     pinMode(PIN_D(p), OUTPUT);
     digitalWrite(PIN_D(p), HIGH);
This appears to have fixed it.

mknerr, 2020-02-19:
I haven't had time to test, and frankly, don't have the skills to fix anything, but just wanted to thank you gentlemen for working this out. Thanks!!

alf45tar, 2020-02-19:
@jimhiggs thank you so much
You are 100% correct. The missing lines can explain all the erroneous beahviours.

alf45tar, 2020-02-19:
Calibration values are saved into EEPROM. Tap dancing i snot rquired at every reboot.

jimhiggs, 2020-02-19:
Great thanks, will give it a try.

jimhiggs, 2020-02-19:
Works great. thanks again.. Now onto my next problem, quite enjoying digging into code.

jimhiggs, 2020-02-20:
On reboot I am getting 3 single presses registered on button 6: Log:
Pedal  1     MIDI         MOMENTARY2   PRESS_1       POLARITY+   NOTE_ON_OFF          6   Channel  1   Pin D25 A36
Pedal  2     MIDI                      PRESS_1       POLARITY+   CONTROL_CHANGE      21   Channel  3
Pedal  3     MIDI                      PRESS_1       POLARITY+   CONTROL_CHANGE      22   Channel  3
Pedal  4     MIDI                      PRESS_1       POLARITY+   CONTROL_CHANGE      23   Channel  3
Pedal  5     MIDI         LADDER       PRESS_1       POLARITY+   NOTE_ON_OFF          6   Channel  1   Pin A32
Pedal  6     MIDI                                    POLARITY+   CONTROL_CHANGE      13   Channel  3
MTC None
Pedal  5   SINGLE PRESS 
NOTE ON.....Note   6.....Velocity   6.....Channel  1
Pedal  5   SINGLE PRESS 
NOTE ON.....Note   6.....Velocity   6.....Channel  1
Pedal  5   SINGLE PRESS 
NOTE ON.....Note   6.....Velocity   6.....Channel  1
I am not touching the pedal.

jimhiggs, 2020-02-20:
I have just installed latest and I also plugged in and configured an analog exp pedal on  4 (also have 6 button on 5 and a boss fs6 on 3). 

When I reboot with monitor I get 
PITCH BEND.....Value  7417.....Channel  3
PITCH BEND.....Value  7417.....Channel  3
Pedal  5   SINGLE PRESS 
NOTE ON.....Note   1.....Velocity   1.....Channel  1
Pedal  5   SINGLE PRESS 
NOTE ON.....Note   1.....Velocity   1.....Channel  1
Pedal  4   input 68 output 120 velocity -0.00
PITCH BEND.....Value  7288.....Channel  3
PITCH BEND.....Value  7288.....Channel  3

jimhiggs, 2020-02-22:
@alf45tar Is the tip ring switch still needed now that you have just one ladder model? Or is it now needed for all ladder? In either case a change to the schematic may be needed.

alf45tar, 2020-02-22:
Tip/ring switch is mandatory for TC Helicon Switch 6 because digital pins cannot be used to as analog input when wifi is on.
If you reversed tip/ring internally you cannot use port 5 for other use.
You use an external cross cable you can reconfigure port 5 for every pedal supported.

jimhiggs, 2020-02-22:
OK but you now have only one solution for all ladders "Ladder". So does the ladder5 shown in the schematic need ring tip reverse or is that still correct?

jimhiggs, 2020-02-22:
Is the Ladder 6 supposed to work with double and long press? Only it only seems to work with single press and it does not respond to 2 quick taps. I have to wait before 2nd press or it is not recognised,  so cannot use for tap tempo etc.

alf45tar, 2020-02-23:
Double press and long press is not yet supported on ladder.

alf45tar, 2020-02-23:
> OK but you now have only one solution for all ladders "Ladder". So does the ladder5 shown in the schematic need ring tip reverse or is that still correct?

Ladder5 do not need ring/tip reverse but it is not reason. 
I shouyd reverse PIN_Dx and PIN_Ax on all the ports.

jimhiggs, 2020-02-23:
OK Thanks for the update

> I should reverse PIN_Dx and PIN_Ax on all the ports.
Sounds like a lot of work.

alf45tar, 2020-02-23:
It is not a lot of work but why?

jimhiggs, 2020-02-23:
No? That is good then.

closed 2020-03-07 by alf45tar