alf45tar/PedalinoMini #67: Default profiles and bank on reset

jimhiggs, 2020-02-01:
When I reset the esp32 and cycle the banks I get A8, B10 and C2 as I cycle the boot switch.
I can't work out what process locks them at a particular profile/bank setting. When in web app live I can set the profile and bank but when I change profile it defaults back to these banks. For a simple pedal expander I would want it to default to A1 on power up. 
Another Q. I have worked out that I can setup a momentary2 button to cycle the banks up and down and I can limit the available banks by setting upper and lower limits. Is there a way to set up a  momentary2 button to cycle profiles and also limit there availability?

jimhiggs, 2020-02-01:
Just an observation, it will allow you to set an upper limit of >10 for a bank change which causes a reboot.

alf45tar, 2020-02-01:
> Is there a way to set up a momentary2 button to cycle profiles and also limit there availability?

Profile cannot be changed using footswitches because it not makes too much sense for me. Which is your requirement?
Profiles are only 3. If you could limit the availability you could limit to 2.

>it will allow you to set an upper limit of >10 for a bank change which causes a reboot.

If you enter a value greater than 10 it will saved as 10. I didin't experenced a reboot in that case.

jimhiggs, 2020-02-01:
Perhaps my understanding is wrong. I viewed profiles as just another 2 sets of 6x10 setups so would want to be able to cycle through them the same as cycling through banks. NP if not available of course. Any help on how to save the default start up profile then?

alf45tar, 2020-02-02:
A profile is a complete configuration of PedalinoMini to be used in a different situation. Each profile can store a different pedal configuration. FOr that reason I think we don't need to switch between profile during a live performance.
If you want more bank you can define here:
https://github.com/alf45tar/PedalinoMini/blob/e0963284ae3477c8c618e4a07456adb0aa44e647/src/Pedalino.h#L20

Remember that there are some memory constraints. If you add more banks reduce the number of profiles.

jimhiggs, 2020-02-02:
Thanks. I understand the use for profiles then and I would agree that there it would not make sense to change profile mid performance. I was thinking that once you then have to open the box to change profile but of course there is always the web i/f or app to change. I suppose I was thinking, once the unit had been completely set up for say 3 different pedal configurations the unit would be stand alone and could be switched between pedal setups without the need to run the app or web i/f if the profile could be switched externally. 

I have just discovered that In order to change the profile I need to change it in pedals on web i/f and then hit save and if I want it to boot at a particular bank I need to change the set the bank in live then go into pedals and save. Would there be any benefit to having  save/set active profile button in live? It is fairly obvious as it is once you know it but It took me a while to figure it out.

alf45tar, 2020-02-03:
BOOT button can be pressed from outside if you add a physical extension.
As alternative re-assign to pin 23.

I could add profile change via footswitch but you should guarantee to assign a profile change to at least one pedal for any profile. Otherwise you stuck in that profile.

jimhiggs, 2020-02-03:
Thank you I will try pin 23.  Yes I understand about the need for a button assignment for each profile. If you think it is a worthwhile enhancement, I would use it, and now you have added the 6 step ladder we have a possible maximum of 36, so plenty spare lol.

jimhiggs, 2020-02-03:
Pin 23 works fine and that would satisfy my needs. Thanks again.

closed 2020-02-03 by jimhiggs

mknerr, 2020-02-03:
@jimhiggs You could also replace the boot micro switch, or even just solder leads of your desired switch to either side of it.

jimhiggs, 2020-02-03:
Yes I did consider that as an option, plus the suggestion made by alf45tar of a physical extension, but pin D23 works does work fine for the ESP32. It is great to be able to define functionality in the web app though and there is a very large number of switch options available. Great design.