alf45tar/PedalinoMini #68: Long Press to send another Control Change/Note

iwbrhwfy-source, 2020-02-08:
Thank you so much, but I think it would be great if 'long press'  send another Control Change/Note, not only send another Midi Value.
ex:
Single Press => CC60
Long Press => CC61

iwbrhwfy-source, 2020-02-08:
And how to change Bank from 1-10 on Pedalion? I set Pedal Mode: Momentary, Pedal Function: Bank+1 but when i press Bank alway to A.10 and only can change via web config
Sorry for my English :D

alf45tar, 2020-02-08:
Check Midi Value 1 and Midi Value 3. They are the lower and upper limit for bank change.

iwbrhwfy-source, 2020-02-08:
> Check Midi Value 1 and Midi Value 3. They are the lower and upper limit for bank change.

Yes i find it :D this is my project :)))
[image]
[image]

alf45tar, 2020-02-08:
I was wrong of course. It is Min and Max the lower and upper limit of bank range.

iwbrhwfy-source, 2020-02-08:
> Thank you so much, but I think it would be great if 'long press' send another Control Change/Note, not only send another Midi Value.
> ex:
> Single Press => CC60
> Long Press => CC61

is it easy to edit?

alf45tar, 2020-02-08:
not so hard

iwbrhwfy-source, 2020-02-08:
> not so hard
Yeah!!
My usbmidi project :D i can select present and turn on off effect with 4 button 
https://youtu.be/mLiJwjWkrs0

iwbrhwfy-source, 2020-02-12:
> not so hard

i just temporary edit code and 4 button can send 8 diffrent CC  :D
[image]
[image]

alf45tar, 2020-07-06:
Solved on latest release 2.0

closed 2020-07-06 by alf45tar