alf45tar/PedalinoMini #69: How to add more noise to potentiometer

iwbrhwfy-source, 2020-02-10:
Please watch my clip 
https://youtu.be/jlJH-D-XVIQ
When i change present, volume pedal in BIAS can update status from analog pedal if it make some noise. But in 0 and 100 it don't noise, bias can't update status

alf45tar, 2020-02-10:
I think I didn't understood the issue.
I see that there is too much noise on pot that it is not filtered by Pedalino.
It looks like that your red push buttons are working ok.

Why you need more noise? Usually it should be filtered. If BIAS do not change presets it not due to noise absence.

iwbrhwfy-source, 2020-02-10:
> I think I didn't understood the issue.
> I see that there is too much noise on pot that it is not filtered by Pedalino.
> It looks like that your red push buttons are working ok.
> 
> Why you need more noise? Usually it should be filtered. If BIAS do not change presets it not due to noise absence.

ahhhh i find it noise because my AA battery low, with USB it clean
ex:
 I save pedal volume on bias fx at 100% (Present A) then I change to present B. When I play I put real expression pedal to 50%
Then if I press Present A volume will from 50 to 100%, but have noise volume still 50% on BIAS
Sorry my bad English, i hope you can understand

iwbrhwfy-source, 2020-02-10:
I just think it will better then noise :)))
went sent midi to change Present, it will send midi of expression pedal. Pedal on bias and real will alway sync

closed 2020-02-15 by iwbrhwfy-source