alf45tar/PedalinoMini #70: Struggling to understand functionality of momentary 2 switch and more

jimhiggs, 2020-02-11:
It would be wonderful if someone who understands how all this works could enlighten me a little. I have been struggling to understand the working of the momentary2 switch setting and was using it with note on/off and am baffled, also with the working of press, double press and long press. It does not do what I expect. I am pretty ignorant on midi coms as well. 
I switched to Control change to see how that worked. What I found confused me. Using the first switch (on portA1)  bank A1 sends the set CC with midi value 1, press it again and it sends the same CC with midi value 2. Alternating between the two. Using the second switch (on PortD1) bank A2 sends the set CC  but it uses the midi value 1 or 2 based on which column switch A1 used. eg if switch 1 used midi value 2 then switch 2 will use the value in its midi value 2.
Is this how it is supposed to work? It may help me understand what is going on with notes. I seem to get switch bounce, ie I press once and get 1, 2 or even 3 midi notes sent, and using the default timings I might get a midi value 1,2 or 3 and it does not seem to correspond well to single double or long press.

iwbrhwfy-source, 2020-02-11:
Momentary 2 only send from D1 pin to display, A1 pin not show in display but it still send midi
defaults: admin, password

jimhiggs, 2020-02-11:
Thanks for the reply. All my observations are from watching the midi output on a midi monitor on pc.  but you puzzle me. I see the midi output on my oled display for both A1 and D1. Will try the login when I get home.

iwbrhwfy-source, 2020-02-11:
> Thanks for the reply. All my observations are from watching the midi output on a midi monitor on pc. but you puzzle me. I see the midi output on my oled display for both A1 and D1. Will try the login when I get home.

sorry it show but can't show diffrent name :D

iwbrhwfy-source, 2020-02-11:
[image]
I found you can change something for Moment 2

jimhiggs, 2020-02-11:
>  sorry it show but can't show diffrent name :D

You can change the name in the banks page for each bank or make them blank to show the midi message.

iwbrhwfy-source, 2020-02-12:
> > sorry it show but can't show diffrent name :D
> 
> You can change the name in the banks page for each bank or make them blank to show the midi message.

thank you so much i just know it @@

jimhiggs, 2020-02-18:
My problem is with my bank up down switches which are cheap chinese push buttons mounted on my pedalino case. I have tried a Boss FS6 on one of my inputs set to Momentary2 and it works fine. The push buttons have terrible bounce. One press will give any number of bank up/downs. Is there anything that can be done to improve this? Save buying new switches of course.

jimhiggs, 2020-02-18:
Actually the FS6  is working normally for bank up and down but I still don't understand why it works the way it does with midi.
I have button A set to C1 CC  1  22   1   2   3
           button B set to C2 CC  1  23 16 17 18
Midi output as seen on monitor:
Press B
B4 16 7F
press  A3 times
B0 16 02
B0 16 02
B0 16 02
press B
B4 16 00
Press A 3 times 
B0 16 01
B0 16 01
B0 16 01
Press B 
B4 16 7F
Press A
B0 16 02
So the output on A depends on the last output on B
ie B value toggles each press but A value changes only after B has changed.

Is this the desired result? I expected A to output Bank A1's midi CC and B to output Bank A2's CC independent of each other?????

jimhiggs, 2020-02-18:
Note on/off pedal 3 Momentary2 Boss FS6. 
Both banks 1 and 2 set up identically.
Button B  (on bank 1)  gives  92 1c 01 on press, 82 1c 02 on release
Button A (on bank 2) gives 92 1C 01 & 82 1C 02 on press, 82 1c 02 on release (every time)
Why are they not the same?

jimhiggs, 2020-02-20:
Closing this because not getting any feedback

closed 2020-02-20 by jimhiggs