alf45tar/PedalinoMini #71: How to show Battery on OLED Display

iwbrhwfy-source, 2020-02-15:
Hello sir
I use 4 AA Battery to VIN of esp32 devkit v1
Please teach me how to show battery status to OLED
My oled is bank at upper right corner

alf45tar, 2020-02-16:
First of all you have to connect the VIN to a available PIN (for example GPIO_NUM_23) using a resistor divider.
Then modifiy Pedalino.h to define the correct BATTERY_PIN and other changes into DisplayOLED.h  as indicated here https://github.com/alf45tar/PedalinoMini/issues/58#issuecomment-579974881.
At the end add -D BATTERY on platformio.ini

iwbrhwfy-source, 2020-02-16:
> First of all you have to connect the VIN to a available PIN (for example GPIO_NUM_23) using a resistor divider.
> Then modifiy Pedalino.h to define the correct BATTERY_PIN and other changes into DisplayOLED.h as indicated here [#58 (comment)](https://github.com/alf45tar/PedalinoMini/issues/58#issuecomment-579974881).
> At the end add -D BATTERY on platformio.ini

And maximum volt to GPIO_NUM_23 sir?

alf45tar, 2020-02-16:
3.3V

iwbrhwfy-source, 2020-02-17:
I used resistor divider same value=10k to gpio23
Then i edit code Battery bat(2000, 3000, GPIO_NUM_23) but i dont know why battery show 0mV :(

iwbrhwfy-source, 2020-02-17:
[image]

alf45tar, 2020-02-18:
Did you disconnected LEFT_PIN and RIGHT_PIN from GPIO_NUM_23 into Pedalino.h?

juliowebhouse, 2020-03-05:
Hi alf45tar,

I followed the instructions to view the battery but it only shows a discharged battery (blinking).

Settings
Battery bat (3000, 5000, GPIO_NUM_23);

#define RIGHT_PIN GPIO_NUM_0
#define CENTER_PIN GPIO_NUM_0
#define LEFT_PIN GPIO_NUM_0

Connections
GPIO_NUM_23 Connected to VIN and Ground using a 10K resistor.

Can you help me? Thanks.

iwbrhwfy-source, 2020-03-05:
> Hi alf45tar,
>
> I followed the instructions to view the battery but it only shows a discharged battery (blinking).
>
> Settings
> Battery bat (3000, 5000, GPIO_NUM_23);
>
> #define RIGHT_PIN GPIO_NUM_0 [...]

I try change pin 23 to another pin exam 34 and it work

alf45tar, 2020-03-05:
The ESP32 has 18 x 12 bits ADC input channels. These are the GPIOs that can be used as ADC and respective channels:

ADC1_CH0 (GPIO 36)
ADC1_CH1 (GPIO 37)
ADC1_CH2 (GPIO 38)
ADC1_CH3 (GPIO 39)
ADC1_CH4 (GPIO 32)
ADC1_CH5 (GPIO 33)
ADC1_CH6 (GPIO 34)
ADC1_CH7 (GPIO 35)
ADC2_CH0 (GPIO 4)
ADC2_CH1 (GPIO 0)
ADC2_CH2 (GPIO 2)
ADC2_CH3 (GPIO 15)
ADC2_CH4 (GPIO 13)
ADC2_CH5 (GPIO 12)
ADC2_CH6 (GPIO 14)
ADC2_CH7 (GPIO 27)
ADC2_CH8 (GPIO 25)
ADC2_CH9 (GPIO 26)

ADC2 pins cannot be used when Wi-Fi is used. So, if you’re using Wi-Fi and you’re having trouble getting the value from an ADC2 GPIO, you may consider using an ADC1 GPIO instead, that should solve your problem.

juliowebhouse, 2020-03-05:
alf45tar,

I set up the GPIO_NUM_36 and now it works, but I'm using a 5v voltage regulator connected to the VIN and a 9v battery and I can't find the right configuration, thanks anyway.

alf45tar, 2020-03-06:
The BATTERY_PIN must be connected to battery output not to voltage regulator if you want measure  the battery voltage.

For more details follow the link
https://github.com/rlogiacco/BatterySense#9v-alkaline-on-5v-mcu

closed 2020-04-08 by alf45tar